oliexdev/openScale #548: 1byOne (CF398BLE)

xxxserxxx, 2020-02-07:
This is the 1byOne that's commonly available it the US on Amazon; at $15 USD it's a low-risk purchase.  I went with this because it appeared from the WiKi that this scale is supported.  I'm trying to troubleshoot measurement issues.

**Describe the bug**
OpenScale sees and pairs with the scale, which shows up as "Health Scale."  It also connects when the scale is activated, and sends a message to "step on the scale with bare feet."  The scale also looks like it's connected (the bluetooth symbol stops flashing, as it does before it was paired).  However, no data is recorded by OpenHealth.

**To Reproduce**
I've
* ensured that users with the same name were registered both in the 1byOne and OpenHealth apps
* eventually filled in all profile data to make sure they matched in both apps
* have tried measuring with 1byOne both installed, and uninstalled

In all cases, OpenHealth behaves the same.

What can I do to see why OpenHealth isn't getting the measurement information from the scale?

I have not tried to reproduce with the latest dev version..

**Expected behavior**
Measurements to be received (from the scale) and recorded.

**Debug log**
[openScale_2020-02-07_05-10.txt](https://github.com/oliexdev/openScale/files/4170461/openScale_2020-02-07_05-10.txt)

xxxserxxx, 2020-02-09:
Paging @oliexdev  -- I'm happy to do diagnostics; can you give me a pointer of what to start with?

oliexdev, 2020-02-27:
What is the OpenHealth app? Mixing and using both app together (openScale with the vendors app) is not a good idea.

But you have a strange error `GATT CONN TERMINATE PEER USER`. I would success using only one app either openScale **or** the vendors app.

xxxserxxx, 2020-02-27:
Initially, openScale would not pair with the scale, which remained in pairing mode.  Attempts to pair the scale using only Android's bluetooth settings failed; it was necessary to install the OpenHealth app to pair the scale.  After pairing the device with the OpenHealth, openScale identified the scale and allowed me to add it, at which time I uninstalled the OpenHealth app.  I have not had the OpenHealth app installed on the phone since after pairing completed.  Since then, openScale sees the scale and asks me to step on the scale.  The scale too acts as if it sees the app (the bluetooth icon stops flashing).  Everything looks like it's working -- it's only that no data is ever recorded by openScale.

To be clear, the log was taken with *only* one app installed on the phone: openScale.  Other than initial pairing, OpenHealth has not been installed.

jimbolaya, 2020-04-22:
I've also experienced data capture issues with this scale. 

It appears to connect fine (the Bluetooth symbol on the app changes to connected and the Bluetooth symbol on the scale flashes). Sometimes I get a result immediately, but many times I'll get no data. I got the best results if I step on the scale immediately after it says "Step on the scale with bare feet", but probably 50% of the time I try to use it, I don't get data unless I retry the whole process a few times, sometimes up to 10 times. 

If I wait long enough after not getting data, it notes that the Bluetooth connection dropped, but that can take several minutes, long after the scale has turned the display off, which I assume means the scale turned off and wasn't transmitting.

xxxserxxx, 2020-04-22:
Interesting.  It connects, and both my phone and the scale say they're connected, but I have never once received data from the scale on my phone.  I may retry several times to see if I can get any response, like you do.

jimbolaya, 2020-04-27:
After some more testing, I found a sequence that gets data over 50% of the time.

1) Make sure the Bluetooth does not attempt to connect to the scale automatically. There's a checkbox in the settings.
2) Make sure the Bluetooth icon in the app is plain white (no connection symbol) or greyed out with a slash through it.
3) Turn on the scale by stepping on it until the display shows up, but otherwise stay off the scale.
4) Click the Bluetooth icon
  - the app will connect to the the scale, then tell you to step on the scale.
5) Step on the scale until the display flashes, indicating it's done measuring, then step off.

You should see a new measurement entry shortly after that. So far trying this, I've only had to repeat the sequence once per measurement session. Many sessions, it just works the first time.

xxxserxxx, 2020-04-28:
You're having more luck than I.  I need to give it a more concerted try, when I get a chance, but I wasn't able to get it to record this morning after I saw your comment.  Does your BT indicator show in the top Android bar when it's doing it's thing?  I see OpenScale saying it's connected/connecting and all that, but Android shows no indication that BT is doing anything.

jimbolaya, 2020-04-28:
I do not get a BT indicator. I'm not sure if that's because this is BT 4.0 BLE and there's a different mode for that. Most IOT bluetooth things use BLE and I believe this scale is no different.

I do get one if I connect a headset or something similar.

One thing that may influence things is 2.4Ghz interference from things like like WiFi or microwave. I sometimes get results if I move my phone closer to the scale (like within a couple feet).

yennsarah, 2020-07-22:
Hej, I got the same scale and debugged to find the error. 
For the check for valid data, the expected length of the byte array is 20, but the scale sends an array with a length of 11. Data gets parsed correctly when I set it to 11, but the app gets a lot of entries, not only 1 measurement with that. ^^'

xxxserxxx, 2020-07-22:
@yennsarah , did you get garbage entries, the same entry repeated, or something else?

yennsarah, 2020-07-23:
I got the same entry repeated, see https://imgur.com/a/IqwJww3

xxxserxxx, 2020-07-23:
I count around 20 entries (at the top) with the same timestamp; some of them have the same data (the first 6 are identical), but it appears to fluctuate. This data looks to me like OpenScale is receiving stream data from the scale -- as if it sends every measurement to the phone as it scans.  The rest of the data looks similar. 

I wonder what would happen if you prevent OpenScale from connecting until after you step off the scale.

yennsarah, 2020-07-27:
Yeah, I also think it's the stream of data. Fluctuation is because I might have moved while measuring to see the values arriving 😅

What exactly do you mean? Just disconnect in the app? 🤔

xxxserxxx, 2020-07-27:
There's a bluetooth bottom at the top (right) of the screen, which -- if you press it -- disconnects the app from the scale.  So,I'm thinking: start disconnected, wait for the scale to settle, then press the bluetooth button to connect.

I may try this. Deleting extra records is probably more trouble than just adding a new entry manually, but it'll be interesting to see if I can get it working.

Did you fork the repo, or just clone it locally?  Did you push your changes into your repo?

xxxserxxx, 2020-07-27:
[On my repo](https://github.com/xxxserxxx/openScale), I've made the change you suggested and started adding code to deal with the duplicates.  It's been many years since I've programmed in Java, and even more since I did anything with Android, so I'm not going to submit this for a PR until it's cleaned up and cross-compatible with other 1byOne scales.

The code in my repository:
1. 1byOne assumes an 11-byte data array (first breakage)
2. Forks a thread to waits for the timestamps to settle
3. If a full 1000ms goes by without a measurement update, the thread writes the record and stops

Improvements needed before a PR is considered:
-  Check for the model number, and only use the new codepath if it matches this model
- I still see a single duplicate on measurement; need to track that down and figure out what causes the dupe
- The 1s window is arbitrary, and could possibly be smaller. Or bigger.  Need to decide if this is sane.

Finally, a far better solution is to see if there's a "final record" marker in the input data. The scale is almost certainly sending a flag that identifies a complete measurement; if we can get this, and all of the thread crap can be deleted.  I'm only exposing this because it *does* work, and -- for me, at least -- something is better than nothing, which is what I have without the change.

yennsarah, 2020-07-28:
> Finally, a far better solution is to see if there's a "final record" marker in the input data. The scale is almost certainly sending a flag that identifies a complete measurement;

That's a really good idea. I'll check for that as soon as I have time again :)

xxxserxxx, 2020-07-28:
I just pushed a new version (to my clone) that trades off complexity for thrashing.

- Threading is removed; the code is simpler
- Duplicates are eliminated -- the previous code produced fewer dupes, but still produced multiple entries; this version does not produce duplicates
- The new code is more thrashy -- it pulles the previous measurement and if it was recorded within 60s and has changed, it updates the record.

I didn't see any unique identifier in the record that says "last measurement." In fact, in tests, I see the same buffer repeated multiple times near the end.  It's still possible that the scale is sending a message saying, "the next message is REALLY the last one", or is sending a "done" message at the end, but in the meantime the new commit records exactly one measurement.

firebovine, 2020-07-31:
Hello

I have just received this scale and have compiled your fork. First -- thanks, great work. I did get data in! I found two bugs, and have one comment.

1. During "Wizard" UI screen (for new install), after selecting this scale, it crashes with a stacktrace (this does not seem to be related to your fork, though....might be upstream):

```
Stack trace: 
java.lang.IllegalArgumentException: ID does not reference a View inside this Activity
	at android.app.Activity.requireViewById(Activity.java:2691)
	at androidx.core.app.ActivityCompat.requireViewById(ActivityCompat.java:363)
	at androidx.navigation.Navigation.findNavController(Navigation.java:58)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$BluetoothDeviceView.onClick(BluetoothSettingsFragment.java:359)
	at android.view.View.performClick(View.java:6669)
	at android.view.View.performClickInternal(View.java:6638)
	at android.view.View.access$3100(View.java:789)
	at android.view.View$PerformClick.run(View.java:26145)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6898)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:537)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
```

2. If a new user tries to use this scale, your fork crashes because `latest.getDateTime()` will return a null object, and `.getTime()` throws an NPE. I manually added a weight and then magic -> got results. We probably just need to check if that's null to be safe.

3. The impedance approach is not very desirable if you want to get scale weights with slippers on (without other metrics). This code just did nothing if I stepped on scale with my slippers on. I don't really have any recommendations for you here, though....too bad there isn't an end code.

Nice work though...the values do show up!

Let me know if you want me to test anything else, or provide additional feedback. :)

xxxserxxx, 2020-07-31:
@firebovine -- thanks for the catches.  It's too bad about the impedance failure cases, but you're right.  The check-and-update method feels so hacky, but I don't see another way of detecting that last measurement.  I pushed a branch on my repo called "check-and-update." If you do run your tests again with that code and it works, let me know and I'll submit a PR. It may not be pretty, but it should be more robust, and it should also handle the other 1byOne models as well.

That `check-and-update` branch is where I'll push a PR from.

joergmlpts, 2020-08-13:
I received my CF398BLE this week. I have captured this communication of the 1byOne app with the scale:

1.  phone to scale: [0xfd, 0x37, 1, 1, 0, 0, 0, 0, 0, 0, xor of previous bytes]
     where the first of the two ones switches the scale to lb, it displays the wight in kg for 0.

2. phone to scale: [0xfd, year >> 8, year & 255, month, day, hour, minute, second]   sets date and time of scale

3. phone to scale: [0xf2, 0] requests data.

The scale responds with multiple notifications of 2, 11, and 18 bytes.

11-byte notifications start with 0xcf and the 11th byte is the xor of the previous 10 bytes. The 10th byte indicates whether the measurement contains all data or only the weight in bytes 4 and 5. 18-byte notifications are identical to 11-byte notifications with the date and time appended in 7 bytes in the order YY-M-D H:M:S.

The 2-byte notification [0xf2, 0] indicates that a stable measurement has been obtained. At that point, the previous 2 measurements appear to always be identical. No more data is received after [0xf2, 0].

After receiving 2-byte notification [0xf2, 0] the 1byOne app sends command [0xf2, 1] to the scale. The scale responds with a 2-byte notification [0xf1 0] followed by numerous 11-byte notifications until it finally turns off. I found no other indication in the Bluetooth traffic that the scale is about to turn off than that two measurements were identical and their 10th byte was zero.

Another observation is that when 11-byte notifications contain the full data and not just the weight, the 10th byte seems to be 0. It is 1 otherwise.

joergmlpts, 2020-08-14:
@xxxserxxx would you mind looking into [this checkin](https://github.com/joergmlpts/openScale/commit/9bdffe89cde0acf306c6bcc2a6e28541a43c6e0a) and pull in any changes you see value in?

That checkin sends a few more of the commands seen in Bluetooth traffic between the scale and the 1ByOne app. There should be value in setting the clock in the scale and parsing time-stamps in the measurements that are received: Supporting time-stamps allows us to step on the scale every day and only occasionally download the measements to the app.

Flags have been introduced that show whether capacitance and date/time are valid. These flags can be checked to e.g. avoid the division by zero fatal.

xxxserxxx, 2020-08-14:
@joergmlpts Will do; I'll ping you all back when I'm done. It'll be a few hours before I can get time to merge this.  Thanks!

xxxserxxx, 2020-08-24:
Sorry for the long delay; IRL intruded.

Using only master on this repo, I'm now getting good results from the scale. It appears that your patch has addressed the issues, and from what I can read, in a cleaner way.  

No changes from my repo are necessary -- as far as I'm concerned, this can be closed. @yennsarah and @firebovine, can either of you confirm that the current head here works for you as well?

yennsarah, 2020-08-25:
@xxxserxxx Master on this repo works fine for me! Great work 👍

xxxserxxx, 2020-08-25:
Seconded.  Closing.  Thanks @joergmlpts !

closed 2020-08-25 by xxxserxxx