oliexdev/openScale #550: Silvercrest SBF76 (Beurer Sanitas?)

GerryGirraffe, 2020-02-14:
Before I start the process of trying to capture the necessary data for a new scale I was wondering if the Silvercrest SBF76 is actually any different to the SBF75 which seems to be supported already?

Happy to try to do the analysis but only if it is not already the same as SBF75. (I don't know what I am doing so it might take me a while, even with the good instructions provided!)

GerryGirraffe, 2020-02-15:
So I have tried to create the data required.

I got stuck at the first step. My phone is not rooted so finding the  btsnoop_hci.log file was nearly impossible. For anyone else having the same problem, I had to generate a Bug Report in the developer menu, then open the Zip file that the Bug Report had generated on my PC and then go to the following folder: 

dumpstate-2020-02-15-15-37-40.zip\FS\data\log\bt

I hope this saves someone else the 90mins that it took me to work out how to find it!

I have a galaxy s9, unrooted.

GerryGirraffe, 2020-02-15:
Well, I think I'm about stuck.

I have finally managed to generate a btsnoop_hci.log but I can't make head nor tail of it.

I can see the scale [Source: ChipseaT_c4:9c:72 (c8:b2:1e:c4:9c:72)] but can't seem to identify any useful information.

I wonder if someone can give me a bit of help?

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/4208803/btsnoop_hci.log)

date 02.15.2020
start time 17:12
weight 81.0
Body Fat 20.6
Water Content 55.2
Muscle 41.1
Bone mass 3.3

boraca, 2020-02-19:
1. Use Wireshark to view the file
2. The protocol is a bit different than SBF75

GerryGirraffe, 2020-02-19:
Thanks for the reply. I tried looking at it with Wireshark, but couldn't understand it. If you could point me towards a more basic tutorial I might have more luck.

Could you check that I have managed to create the right type of file with the right kind of data in it? It would be good to know that I got that bit right at least.

Thanks

GerryGirraffe, 2020-02-25:
Well after a few weeks with the scales I've decided that they are just not worth the trouble. Variation between readings of 300g are typical, but I've also seen 500g differences between repeat weighings. That and I discover they weigh about 800g-1kg below a set of calibrated scales. So I'll be taking them back for a refund.

closed 2020-02-27 by oliexdev

raspispass, 2022-07-14:
The SBF76 seems to be an equivalent to the supported SBF77/BF950. By just adding the deviceName comparison for SBF76 in BluetoothFactory.java all measurements are taken correctly and even new users can be added:

       if (deviceName.equals("SBF77") || deviceName.equals("SBF76") || deviceName.equals("BF950")) {
            return new BluetoothBeurerBF950(context, deviceName);
        }

I would be happy if @oliexdev would add the change to the master branch.