oliexdev/openScale #552: Bug with Beurer BF700

maniackcrudelis, 2020-02-23:
**Describe the bug**
*The app seems to work and show the weight, received from the scale. But nothing is stored.
I have to try many times to finally have my data stored.*

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): ???
*I use the version 2.1.2(47), from FDroid.*

**Expected behavior**
*I expect data to be stored when I connect the app to the scale.*

**Additional context**
Add any other context about the problem here.

**Debug log**
```
2020-02-23 23:12:01.222 Debug [2] AboutPreferences: Debug log enabled, openScale v2.1.2 (47), SDK 28, Sony I4113
2020-02-23 23:12:01.226 Debug [2] AboutPreferences: Selected user id(1) name(***) birthday(***) age(***) body height(***) scale unit(kg) gender(***) initial weight(***) goal weight(***) goal date(***) measure unt(cm) activity level(***)
2020-02-23 23:12:04.921 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: ***
2020-02-23 23:12:06.889 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: ***
2020-02-23 23:12:07.132 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: ***
2020-02-23 23:12:08.261 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: ***
2020-02-23 23:12:09.359 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: ***
2020-02-23 23:12:10.514 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: ***
2020-02-23 23:12:11.612 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: ***
2020-02-23 23:12:13.060 Info [2] BluetoothBeurerSanitas: Active measurement, stable weight: ***
2020-02-23 23:12:14.465 Info [2] BluetoothBeurerSanitas: Active measurement, stable weight: ***
2020-02-23 23:12:19.631 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 01] on 0xffe1
2020-02-23 23:12:19.659 Debug [2] BluetoothPeripheral: writing <e7f1590301> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-02-23 23:12:19.767 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-02-23 23:12:19.768 Debug [2] BluetoothCommunication: Step Nr 8
2020-02-23 23:12:19.768 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2020-02-23 23:12:19.869 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 02] on 0xffe1
2020-02-23 23:12:19.886 Debug [2] BluetoothPeripheral: writing <e7f1590302> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-02-23 23:12:20.008 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 02> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-02-23 23:12:20.009 Debug [2] BluetoothCommunication: Step Nr 8
2020-02-23 23:12:20.009 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2020-02-23 23:12:20.124 Error [2] BluetoothBeurerSanitas: java.lang.NullPointerException: Attempt to read from field 'int com.health.openscale.core.bluetooth.BluetoothBeurerSanitas$RemoteUser.localUserId' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.processMeasurementData(BluetoothBeurerSanitas.java:350)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.processMeasurement(BluetoothBeurerSanitas.java:401)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.onBluetoothNotify(BluetoothBeurerSanitas.java:282)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:343)
	at com.welie.blessed.BluetoothPeripheral$1$7.run(BluetoothPeripheral.java:534)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:280)
	at android.app.ActivityThread.main(ActivityThread.java:6744)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
```

oliexdev, 2020-02-27:
There is something wrong with the user. Could you do a full reset on the scale and in the app and initialize the scale again?

maniackcrudelis, 2020-03-02:
Sorry for the late answer, I did some tests
First I reset the scale only and it seems to work, but only once actually...
After that the same behavior happened.

So I did reset both the scale and the app.
But since then, that's even worse since the scale doesn't recognize me anymore and there's still no data saved into the app

```
2020-03-02 20:05:19.058 Debug [2] MainActivity: Bluetooth connection successful established
2020-03-02 20:05:19.240 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2020-03-02 20:05:19.241 Debug [2] BluetoothCommunication: Resume machine state
2020-03-02 20:05:19.242 Debug [2] BluetoothCommunication: Step Nr 1
2020-03-02 20:05:19.243 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2020-03-02 20:05:19.243 Debug [2] BluetoothCommunication: Stop machine state
2020-03-02 20:05:19.247 Debug [2] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:19.329 Debug [2] BluetoothCommunication: SUCCESS: Writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:19.420 Debug [2] BluetoothBeurerSanitas: Got init ack from scale; scale is ready
2020-03-02 20:05:19.422 Debug [2] BluetoothCommunication: Resume machine state
2020-03-02 20:05:19.422 Debug [2] BluetoothCommunication: Step Nr 2
2020-03-02 20:05:19.423 Debug [2] BluetoothCommunication: Invoke write bytes [E9 5E 5D 58 EF] on 0xffe1
2020-03-02 20:05:19.424 Debug [2] BluetoothCommunication: Step Nr 3
2020-03-02 20:05:19.425 Debug [2] BluetoothCommunication: Invoke write bytes [E7 4F 00 00 00 00 00 00 00 00] on 0xffe1
2020-03-02 20:05:19.426 Debug [2] BluetoothCommunication: Stop machine state
2020-03-02 20:05:19.429 Debug [2] BluetoothPeripheral: writing <e95e5d58ef> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:19.509 Debug [2] BluetoothCommunication: SUCCESS: Writing <E9 5E 5D 58 EF> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:19.513 Debug [2] BluetoothPeripheral: writing <e74f0000000000000000> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:19.599 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 4F 00 00 00 00 00 00 00 00> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.050 Debug [2] BluetoothBeurerSanitas: Battery level: 100; threshold: weight=2.00, body fat=2.00; unit: 1; requested user: exists=false, has reference weight=false, has measurement=false; scale version: 5
2020-03-02 20:05:20.052 Debug [2] BluetoothCommunication: Resume machine state
2020-03-02 20:05:20.054 Debug [2] BluetoothCommunication: Step Nr 4
2020-03-02 20:05:20.055 Debug [2] BluetoothCommunication: Invoke write bytes [E7 33] on 0xffe1
2020-03-02 20:05:20.056 Debug [2] BluetoothCommunication: Stop machine state
2020-03-02 20:05:20.059 Debug [2] BluetoothPeripheral: writing <e733> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.139 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 33> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.320 Debug [2] BluetoothBeurerSanitas: Have 1 users (max is 8)
2020-03-02 20:05:20.409 Debug [2] BluetoothBeurerSanitas: Received user 1/1: DOR (1984)
2020-03-02 20:05:20.410 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 34 01 01] on 0xffe1
2020-03-02 20:05:20.417 Debug [2] BluetoothBeurerSanitas: Remote user DOR (0x65) is local user Dorian (1)
2020-03-02 20:05:20.420 Debug [2] BluetoothCommunication: Resume machine state
2020-03-02 20:05:20.421 Debug [2] BluetoothCommunication: Step Nr 5
2020-03-02 20:05:20.422 Debug [2] BluetoothBeurerSanitas: Request saved measurements for DOR
2020-03-02 20:05:20.424 Debug [2] BluetoothCommunication: Invoke write bytes [E7 41 00 00 00 00 00 00 00 65] on 0xffe1
2020-03-02 20:05:20.425 Debug [2] BluetoothCommunication: Stop machine state
2020-03-02 20:05:20.429 Debug [2] BluetoothPeripheral: writing <e7f1340101> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.499 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 34 01 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.503 Debug [2] BluetoothPeripheral: writing <e7410000000000000065> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.590 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 41 00 00 00 00 00 00 00 65> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.679 Debug [2] BluetoothBeurerSanitas: No saved measurements found for user DOR
2020-03-02 20:05:20.680 Debug [2] BluetoothCommunication: Jump next to step nr 5
2020-03-02 20:05:20.681 Debug [2] BluetoothCommunication: Resume machine state
2020-03-02 20:05:20.682 Debug [2] BluetoothCommunication: Step Nr 5
2020-03-02 20:05:20.683 Debug [2] BluetoothCommunication: Step Nr 6
2020-03-02 20:05:20.684 Debug [2] BluetoothCommunication: Step Nr 7
2020-03-02 20:05:20.686 Debug [2] BluetoothCommunication: Invoke write bytes [E7 36 00 00 00 00 00 00 00 65] on 0xffe1
2020-03-02 20:05:20.687 Debug [2] BluetoothCommunication: Stop machine state
2020-03-02 20:05:20.690 Debug [2] BluetoothPeripheral: writing <e7360000000000000065> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.770 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 36 00 00 00 00 00 00 00 65> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-03-02 20:05:20.859 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 29.85
[... cut ...]
	at com.welie.blessed.BluetoothPeripheral$1$7.run(BluetoothPeripheral.java:534)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:280)
	at android.app.ActivityThread.main(ActivityThread.java:6744)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
```

oliexdev, 2020-03-08:
Looks fine until the last error with the localUserId. 

Could you reset everything (app and scale) and register again while your logging in openScale? I want to check if the user creation works fine and please post the full log file.

maniackcrudelis, 2020-03-08:
Ok, so I reset both scale and app, recreate an user, with default values and it worked...
I reset again the app (not the scale) and recreate my common user, still working...

Finally I didn't reproduce for now. Last time I did it, it worked at first and failed later. We'll see...
Anyway, here the log of my last reset:
https://crudelis.fr/zerobin/?8c2c2b2bbe9db8e2#2PdDcouN9tfSTlmPe23M8npOWSMhZ4bvNUGIOeKANi4=

oliexdev, 2020-03-10:
Did you use an empty user name in openScale or did you delete it in the log file? Also how many users did you have in openScale? More than one?

maniackcrudelis, 2020-03-10:
Into my last log, I didn't use an empty user nor remove it from the log.
But after the reset, I didn't create an user at first, to enable the log. Then, I went back to the user list to create the user.

I only had 1 user all long.

oliexdev, 2020-03-10:
ok then your log file makes sense because in the log file your user name were empty and the user id mismatched with the scale user id.

I will close this issue until your problem occurs again.

closed 2020-03-10 by oliexdev

maniackcrudelis, 2020-03-10:
You won't have to wait long before the problem occurs again...
Just used it and again, the scale does not recognize the user. And so the app does not store any data.
Here the log:
https://crudelis.fr/zerobin/?5908914afd717e5c#MEIqtc7BG0JOIcZN+HbKFv23akslf2NVyKmZGW4Kb+A=

Note that this time I didn't reset anything, just used the app, connected to the scale and used it.

oliexdev, 2020-03-11:
It seems openScale wants to create again the user on the scale!? I am not sure but did you connect to the scale in the initialization process without an openScale user?

Your log files are very strange....

In your first log:
1. At first no user is created in openScale
2. Even if you reset your scale, openScale found a user on the scale `Received user 1/1: USE (2000)` So I assume your scale was not fully reset.
3. You created a user a user on the scale `Create user: Dorian`

second log file:
1. no users were found on the scale
2. openScale want to create a user on the scale but won't success (because there is already a user)

Are you using in parallel the vendors app or another app?
It is again something wrong with yours user creation process?
Did your scale work with the vendors app?

maniackcrudelis, 2020-03-11:
First, to answer yours questions:
I never used the vendor app. So I can't tell if it works with it or not.
Not using any other app either.
When creating the user, nothing seems to go wrong.

Ok, I reset everything again, and here step by step what I'm doing:

- I start the app, which ask me to create an user.
- Once the user created, I turn on the logging.
- Turn on Bluetooth and GPS
- Add the scale into the Bluetooth menu of the app.
- Connect the app to the scale (with the Bluetooth button on the upper side of the app)
- Wait for the app to connect and use the scale.
- Everything's ok.
Note that it usually works until then. The problem is more about what's going on after that.

After 5 minutes, Bluetooth and GPS turned off. I retried:
- Start the app, Bluetooth turned on by the app.
- I wait for the app to connect to the scale.
(I quickly saw the error message about the max number of concurrent users)
- The scale recognize the user and works as expected.
- Nothing stored into the app, but I expect it's because of the 5 minutes only since the last measurement.

I'll retry this evening or tomorrow but don't expect it to work, since it always works fine at first but not after.

oliexdev, 2020-03-11:
I'm at a bit of a loss. If you have reset the scale, you shouldn't see the max number of concurrent users error!? Just to be sure, did you really reset the scale? You have to push the reset button on the scale until a del message occurred on the scale display.

maniackcrudelis, 2020-03-11:
That's indeed what I do.
Also there's only one button on this scale. So I turned it on, hit the button until it says "del"
Actually, the scale says that since the beginning, even if I never created more than 2 different users.

maniackcrudelis, 2020-03-12:
Same behavior today than previously, the app connect to the scale, says the same error message about max number of concurrent users.
Then the scale does the job, but nothing is stored into the app.

https://crudelis.fr/zerobin/?74cda247b6c0b788#fSdMjog3TuIdRpj4mK0v3AXWVwwX4geqf+FJsTHzfC4=

roerch, 2020-03-13:
I am also affected by this bug. Same scale, same problem. Initialization has been done with this app, the vendors app was never installed. After two measurements I get the above error about maximum number of users on scale. Having a very quick look at the code it seems to me like the user defined in the app is never found on the scale and every time a new user is created (maybe it is as simple as a 'off by one' bug.
Is there a way to debug this process of querying for the user, like logging it to the debug log?

maniackcrudelis, 2020-03-19:
@oliexdev would you mind considering reopen this issue, as it's not solved.
Also if there's anything else I could do to help find what the problem is, let me know. I would really like to eventually use the scale and this app.

roerch, 2020-04-26:
Just as a follow-up here: In Issue #565 there is a reference to a comment in the Play Store with a working hint. Turning on the GPS before starting the app and connecting to the scale fixes the problem for me. Seems to be a permission issue.

gnaghi, 2020-05-09:
I have the same problem here.
Openscale recorded the first weight, but the day after, I have the same com.health.openscale.core.bluetooth.BluetoothBeurerSanitas$RemoteUser.localUserId error.

gnaghi, 2020-10-09:
@oliexdev : I have the same bug, with the latest version (openScale v2.3.1).

`2020-10-09 10:13:56.436 Debug [2] AboutPreferences: Debug log enabled, openScale v2.3.1 (50), SDK 26, samsung XXX
2020-10-09 10:13:56.440 Debug [2] AboutPreferences: Selected user id(1) name(XXX) birthday(XXX) age(XX) body height(XXX,00) scale unit(kg) gender(male) initial weight(XX,00) goal weight(XX,00) goal date(XXX 00:00:00 GMT+01:00 2020) measure unt(cm) activity level(2) assisted weighing(false)
2020-10-09 10:14:07.976 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: XXX
2020-10-09 10:14:08.076 Debug [2] MainActivity: Bluetooth scale message: En train de mesurer le poids : XXX
2020-10-09 10:14:14.768 Info [2] BluetoothBeurerSanitas: Active measurement, stable weight: XXX
2020-10-09 10:14:15.919 Info [2] BluetoothBeurerSanitas: Active measurement, stable weight: XXX
2020-10-09 10:14:21.226 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 01] on 0xffe1
2020-10-09 10:14:21.241 Debug [2] BluetoothPeripheral: writing <e7f1590301> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-10-09 10:14:21.318 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-10-09 10:14:21.321 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 02] on 0xffe1
2020-10-09 10:14:21.326 Debug [2] BluetoothPeripheral: writing <e7f1590302> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-10-09 10:14:21.417 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 02> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-10-09 10:14:21.425 Error [2] BluetoothBeurerSanitas: java.lang.NullPointerException: Attempt to read from field 'int com.health.openscale.core.bluetooth.BluetoothBeurerSanitas$RemoteUser.localUserId' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.processMeasurementData(BluetoothBeurerSanitas.java:350)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.processMeasurement(BluetoothBeurerSanitas.java:401)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.onBluetoothNotify(BluetoothBeurerSanitas.java:282)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:358)
	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:438)
	at android.os.Handler.handleCallback(Handler.java:789)
	at android.os.Handler.dispatchMessage(Handler.java:98)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6944)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)`

If I try multiple time (at least 6 or 7 times), without changing anything, the record is stored.
The bug occured today, but the software has been working without error for 2 months, and I didn’t change anything on my phone.