oliexdev/openScale #553: Calendar integration

alexanderadam, 2020-02-25:
It's a luxury feature but I thought it would be nice if openScale could also add entries to a calendar.
Each title could contain a summary and some details in the description.

oliexdev, 2020-02-27:
What should the calendar entry contain in the summary and in the detailed description?

alexanderadam, 2020-02-27:
Maybe something like

**Title**: `Weight on May 23, 2023 at 12:12 p.m .: xxkg`

**Description** (maybe just showing the details that are selected for viewing in the app):

```
Body mass index (BMI): xxx
Weight difference: xx.x
Body fat: xx.x
…
```

oliexdev, 2020-03-08:
Wouldn't you spam your calendar with these information? I think this feature is very special, I see no really benefit from it.

closed 2020-03-08 by oliexdev