oliexdev/openScale #556: Mass Array of Scale and it must has feature.

seasiri, 2020-03-08:
**Hardware prepared:**
x1 Android tablet install with wall mount (for receive data from 60 scales)
x60 unit Xiaomi Scale V2 
x1 Openscale app pro (purchased)
x1 Openscale Sync (purchased)
**Thing I am trying to do:**

> Object is on scale 24/7. So the weight is always can be capture.

x60 Xiaomi V2 capture data every 10 min.  
  ------ (can it capture every 10 min. or every hours. what is optimize time for battery)
Store in android tablet (advice please may be better idea)
sync to CloudMqtt  (not problem)

I want battery to last 5 month with this task. Is it possible?
Can it capture every 10min while object always on scale?
Can openscale app open 60 connection of Xiaomi Scale 2?

Thank you for kind help in advance,
@oliexdev  @Peterkn2001

oliexdev, 2020-03-08:
I have never seen body weight scales that weight constantly. OpenScale can also connect to only one scale at a time. You have to build your own scale to fulfill your requirement. What do you want to weight with so many scales?

seasiri, 2020-03-09:
There is reason, for constant measuring for example. Measuring screw stock. Screw box place on Xiaomi v2 scale. I think you know where am I going with this. convert raw weight into good quantity. 

Problem currently, how do I turn on scale when there is object constantly on the scale. Do I need sonoff relay to turn on xiaomi device and bypass power inside xiaomi possibility is endless and there should be one best answer to do this in the most efficient manner. 

How do I manage on and off with constant measuring? 
Is there a work around for unforseen problem that might surprise?
How do I send weight data every 10 min to mysql database?

 I am open developer to propose and discuss about possibility of this with of course money so we can eat.

oliexdev, 2020-03-10:
Maybe you can trigger the xiaomi v2 scale with a microcontroller. My custom scale has a pressure plate which starts the measurement (the round thing in [this picture](https://github.com/oliexdev/openScale/raw/master/docs/custom_scale/circuit_board_back_raw.JPG))
I don't know how its works with the xiaomi mi scale. You have to look. But I think for your case  it would be easier and cheaper to buy some raw pressure weight sensors and some microcontrollers and hack it together. Then you have the full control and you can do a constant measurement if you want.

seasiri, 2020-03-11:
I will try to hack Xiaomi scale 2 with microcontroler first and then I will report back to this topic so that people can use my experiment as information in the future. 
Great. 
the reason im still pushing toward xiaomi scale is because it housing and material is totally fine to touch, it cost only 15 dollar here. nothing can beat this price even do assemble load cell and micro  from ground up also need housing.
.
therefore I need some love and observer to participate with me Thank you admin

oliexdev, 2020-03-11:
If they are so cheap and you are ok with it then you can try to do it with the xiaomi mi scales.
If you have disassemble the scale you can post some picture here or create a new GitHub project in your repository. Maybe somebody are interested in your idea.

seasiri, 2020-03-11:
im no expert in github. i can try to fork from you if that right? then keep update. i need your kindly mentor love.

oliexdev, 2020-03-11:
I would recommend to do a new GitHub project and write down yours experiences in a GitHub wiki.

oliexdev, 2020-03-30:
I will close this issue, you can keep us here informed about your project.

closed 2020-03-30 by oliexdev

daleyjem, 2023-02-07:
@oliexdev  Did you make any progress on this?

tojo99, 2023-12-29:
> If they are so cheap and you are ok with it then you can try to do it with the xiaomi mi scales.
> If you have disassemble the scale you can post some picture here or create a new GitHub project in your repository. Maybe somebody are interested in your idea.

> **Hardware prepared:** x1 Android tablet install with wall mount (for receive data from 60 scales) x60 unit Xiaomi Scale V2 x1 Openscale app pro (purchased) x1 Openscale Sync (purchased) **Three Thing I am trying to do:**
> 
> > Object is on scale 24/7. So the weight is always can be capture.
> 
> 1. x60 Xiaomi V2 capture data every 10 min.
>    ------ (can it capture every 10 min. or every hours. what is optimize time for battery)
> 2. Store in android tablet (advice please may be better idea)
> 3. Sync to CloudMqtt  (not problem)
> 
> I want battery to last 5 month with this task. Is it possible? Can it capture every 10min while object always on scale? Can openscale app open 60 connection of Xiaomi Scale 2?
> 
> Thank you for kind help in advance, @oliexdev @Peterkn2001
> 
> Reference unsolve problem like my question: [#96 (comment)](https://github.com/oliexdev/openScale/issues/96#issuecomment-579612765)

@seasiri Hi, did you try switchbot for wakeup scale?