oliexdev/openScale #557: Crash when saving table row.

william-sy, 2020-03-08:
**bug**
When editing a table row and pressing save the app crashes (attach a different user to row)

**To Reproduce**
Nokia 7 plus, Android 10
1. Open app 
2. menu
3. table
4. open row you want to edit
5. press pencil icon
6. change name
7. press ok
8. press save icon 
App should now crash

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): no just the app store version.

**Expected behavior**
Save the row and not crash.

**Additional context**
N/A

**Debug log**
Build version: 2.2.0-pro
Current date: 2020-03-08 11:07:06
Device: HMD Global Nokia 7 plus
 
Stack trace:  
android.database.sqlite.SQLiteConstraintException: UNIQUE constraint failed: scaleMeasurements.userId, scaleMeasurements.datetime (code 2067 SQLITE_CONSTRAINT_UNIQUE)
at android.database.sqlite.SQLiteConnection.nativeExecuteForChangedRowCount(Native Method)
at android.database.sqlite.SQLiteConnection.executeForChangedRowCount(SQLiteConnection.java:831)
at android.database.sqlite.SQLiteSession.executeForChangedRowCount(SQLiteSession.java:756)
at android.database.sqlite.SQLiteStatement.executeUpdateDelete(SQLiteStatement.java:66)
at androidx.sqlite.db.framework.FrameworkSQLiteStatement.executeUpdateDelete(FrameworkSQLiteStatement.java:46)
at androidx.room.EntityDeletionOrUpdateAdapter.handle(EntityDeletionOrUpdateAdapter.java:70)
at com.health.openscale.core.database.ScaleMeasurementDAO_Impl.update(ScaleMeasurementDAO_Impl.java:171)
at com.health.openscale.core.OpenScale.updateScaleData(OpenScale.java:382)
at com.health.openscale.gui.activities.DataEntryActivity.saveScaleData(DataEntryActivity.java:395)
at com.health.openscale.gui.activities.DataEntryActivity.onOptionsItemSelected(DataEntryActivity.java:218)
at android.app.Activity.onMenuItemSelected(Activity.java:4137)
at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:384)
at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:219)
at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:64)
at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:207)
at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:840)
at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:991)
at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:981)
at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:151)
at android.view.View.performClick(View.java:7125)
at android.view.View.performClickInternal(View.java:7102)
at android.view.View.access$3500(View.java:801)
at android.view.View$PerformClick.run(View.java:27340)
at android.os.Handler.handleCallback(Handler.java:883)
at android.os.Handler.dispatchMessage(Handler.java:100)
at android.os.Looper.loop(Looper.java:214)
at android.app.ActivityThread.main(ActivityThread.java:7397)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)

oliexdev, 2020-03-08:
what do you mean by step 6? What is name?

william-sy, 2020-03-15:
Attached a screen shot, Changing the name here from user1 to user2 and then pressing save will crash the app.
[image]

oliexdev, 2020-03-24:
I can't reproduce this behaviour.

closed 2020-03-30 by oliexdev

william-sy, 2020-12-31:
I would like to reopen this as I have still got this problem.

Now I just have the root cause of this :) 

Once me and my partner have been on the scale I run the import on my phone.
1: Open app
2: Connect to scale
3: Download data

This is the result in the DataBase: 
[image]
Row 2 and 3 are duplicate 1 row for me and one row for my partner. 
Here without me trying to change it: 
[image]

It would be nice to clean this up in the app. Or giving me the option which row to keep instead of crashing, or any other sollution. 

Now to fix it I do the following: 
1: Export data
2: remove data from app 
3: Use sqlite editor to remove duplicate row
4: Reimport data

I have no idea if the scale sends 2 rows of data or the app does this either way It would be nice to fix this somehow.