oliexdev/openScale #559: Decrease initial weight measurement

ofeksr, 2020-03-14:
Hello,
I'm part of a group who built a scale adaption, for wheelchair users to measure their weight at home, you can see more info here: [Upscale](https://tomglobal.org/project?id=5d5a7a8dc37c3b69559c3b6d)
Since our users can't directly see their weight, they use mobile app.

The problem is that they need to subtract manually the scale adaption and wheelchair weights -- we want to have a mobile app with a setting feature that will allow us to decrease the initial weight by default.

I can imagine a checkbox on first setup screen (and in settings menu also) with "Wheelchair User?", then adding "Wheelchair Weight" and "Scale Adaption Weight" for auto subtraction.

Is it possible to add this feature? and if not, i'll be happy to dive in and try developing it by myself.
Thank you.

oliexdev, 2020-03-24:
I guess its possible the problem if we introduce a new "Scale Adaption Weight" all derived from weight values (e.g. BMI and so on) are wrong. So we need to directly subtract the internal weight value. 

I guess it would also a benefit for users with an amputate leg / arm, wouldn't it?

I am not a medical expert but are the calculations for BMI, TDEE, BMR for wheelchair users or users with amputate arm / leg correct?

ofeksr, 2020-03-24:
I'v googled it and there are ways to measure them correctly, you can see here for example:
- [BMI Calculator for Amputees](https://www.amputee-coalition.org/limb-loss-resource-center/resources-filtered/resources-by-topic/healthy-living/)
- [ Adjust Calorie Needs for an Amputation](https://www.livestrong.com/article/505596-how-to-adjust-calorie-needs-for-an-amputation/)

But if it complicating things - we'll be more than happy with only regular weight measurement and its fine to disable the other calculations (if its possible and okay of course.)

Thanks for you reply!

oliexdev, 2020-03-24:
I created a new branch please check it out and test it. I called it "subtraction weight" in the user settings which are subtracted if you enter a new weight. Do you have a better word for it?
I also want to add a field for Amputees which are predefined like your mention BMI Calculator but I am a bit loss how I calculate the subtraction? I have found the following table https://aprpros.com/true-weight-amputees/ any ideas? Currently the weight is subtracted with "subtraction weight" if you enter a new weight.

ofeksr, 2020-03-24:
Great ! 
Just can't manage to generate apk.. 😄

oliexdev, 2020-03-24:
you can use the Travis build @ https://github.com/oliexdev/openScale/releases/tag/travis-dev_WheelChairFeature-build

oliexdev, 2020-04-03:
@ofeksr any news or not interested anymore?

ofeksr, 2020-04-03:
Waiting for Travis to generate apk...?

oliexdev, 2020-04-03:
maybe generate the apk by yourself?

ofeksr, 2020-04-03:
Trying... 
First i'v seen this:
[image]
so i copied androidmanifest to root dir.
and then i run after connecting my device, and got this:
```
Error:Internal error: (java.lang.ClassNotFoundException) com.google.wireless.android.sdk.stats.IntellijIndexingStats$Index
java.lang.ClassNotFoundException: com.google.wireless.android.sdk.stats.IntellijIndexingStats$Index
	at java.net.URLClassLoader.findClass(URLClassLoader.java:382)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at com.intellij.util.indexing.counters.IndexCounters.<clinit>(IndexCounters.java:34)
	at com.intellij.util.indexing.impl.MapReduceIndex.<init>(MapReduceIndex.java:94)
	at com.intellij.util.indexing.impl.MapReduceIndex.<init>(MapReduceIndex.java:110)
	at org.jetbrains.jps.backwardRefs.index.CompilerReferenceIndex$CompilerMapReduceIndex.<init>(CompilerReferenceIndex.java:248)
	at org.jetbrains.jps.backwardRefs.index.CompilerReferenceIndex.<init>(CompilerReferenceIndex.java:84)
	at org.jetbrains.jps.backwardRefs.JavaCompilerBackwardReferenceIndex.<init>(JavaCompilerBackwardReferenceIndex.java:12)
	at org.jetbrains.jps.backwardRefs.JavaBackwardReferenceIndexWriter.initialize(JavaBackwardReferenceIndexWriter.java:80)
	at org.jetbrains.jps.incremental.java.JavaBuilder.buildStarted(JavaBuilder.java:149)
	at org.jetbrains.jps.incremental.IncProjectBuilder.runBuild(IncProjectBuilder.java:359)
	at org.jetbrains.jps.incremental.IncProjectBuilder.build(IncProjectBuilder.java:178)
	at org.jetbrains.jps.cmdline.BuildRunner.runBuild(BuildRunner.java:139)
	at org.jetbrains.jps.cmdline.BuildSession.runBuild(BuildSession.java:288)
	at org.jetbrains.jps.cmdline.BuildSession.run(BuildSession.java:121)
	at org.jetbrains.jps.cmdline.BuildMain$MyMessageHandler.lambda$channelRead0$0(BuildMain.java:228)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

```

closed 2020-06-18 by oliexdev