oliexdev/openScale #562: Vigorun support missing for devices named `YG`

thebabush, 2020-03-28:
**Describe the bug**

I have a brand new Vigorun scale but openScale's UI says it's not supported.
Looking at the code, it seems that you pattern match the bluetooth device name.
In my case the name `YG` and it is not one of the two you use for Vigorun-like devices.

Anyway, couldn't help myself from working on it this afternoon of COVID quarantine.
I made a [partial implementation of Vigorun protocol in python](https://gist.github.com/thebabush/7c0c4582880114bfa0edadef0cd281fe).

My scale is actually simpler than what the code supports, so I only tested the weight / impedence (adc) and the values that are derived from that.

**To Reproduce**
Steps to reproduce the behavior:

1. Buy a Vigorun
2. Scan in openScale
3. Not supported

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 87775f6

**Expected behavior**
The scale should be supported.

**Other**

Sorry I didn't make a PR, but I don't have an Android dev setup at the moment and I just wanted to give some info.

Also, I have some (non bluetooth-)reversing skills so I could actually do more, but I dunno your policy on that... Being a gray area and all of that.

Let me know what you think.

thebabush, 2020-03-29:
Ok turns out the current implementation is working out of the box (the things I reversed are for the "real time" results, not for the final measurement).

See #563

oliexdev, 2020-03-30:
thanks, I merged your PR. Did you reverse engineer the fat, water and so on for the MGB scales in your python implementation? Did it work?

thebabush, 2020-03-30:
I was focusing on the 8-byte long messages and on my scale they are basically sending weight and then `adc` which is some sort of inter-leg impedence I suppose. It's single valued so, given also the name, I suppose it's the analog voltage of some sensor.

On the other hand, openScale uses 20-bytes long messages. that seems like some sort of historical data.

The app supports other messages that I haven't seen, so I guess they have multiple firmwares/scales.

Anyway, the historical values might also contain the stuff you want (you only use some bytes) but I'm not completely sure. I can spend some time to look at it if you want. Just give me the values you would like to have.

Last but not least, they have formulas to derive several indicators from age/height/sex/weight/adc. I don't really believe it's very scientific lol, but they are not too difficult to reverse (in fact, I did it in my python scripts, just not for everything). From the look of it, it's standard "fit this simple function to approximate our data" kind of formulas. They are similar to what you can find on papers online, but the constants are different ofc.

oliexdev, 2020-03-31:
Well, I don't own this type of scale it was a PR #90 by @DreamNik, so I can't test it.
It would be great if you could send a PR with the missing body measurements for these scales. The implementation is in `BluetoothMGB` class. In fact we need to find the adc / impedance value in the bytes and then see if your formulas give us the same results as the vendors app.