oliexdev/openScale #564: Wiki Xiaomi Body Composition Scale 2

BeMacized, 2020-04-04:
This issue isn't related to the project as much as it is about the Wiki.

Recently I purchased a "Xiaomi Mi Body Composition Scale 2". 

As I was hoping to do some more stuff with it, I have been trying to parse the info from its Bluetooth advertisement data.

Your [wiki page](https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale) on the v1 version of this device is one of the few sources I could find which provided helpful info (Thank you!!). I discovered that the v2's format is a bit different, but I managed to parse some of the data it advertises.

This is what I could find out:
| Bytes | Type   | Endianness | Function                                                                                                                                                                                                                                       |
|-------|--------|------------|------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| 0-1   | -      | -          | Various status flags<br> - Bit 0: ?<br> - Bit 1: Measurement Status (1: Complete, 0: Measuring)<br> - Bit 2-4: ?<br> - Bit 5: Weight Stabilized<br> - Bit 6: ?<br> - Bit 7: Weight Removed<br> - Bit 8: Unit (1: LBS, 0: KG)<br> - Bit 9-15: ? |
| 2-3   | Uint16 | Little     | Year                                                                                                                                                                                                                                           |
| 4     | Uint8  | -          | Month                                                                                                                                                                                                                                          |
| 5     | Uint8  | -          | Day                                                                                                                                                                                                                                            |
| 6     | Uint8  | -          | Hour                                                                                                                                                                                                                                           |
| 7     | Uint8  | -          | Minute                                                                                                                                                                                                                                         |
| 8     | Uint8  | -          | Second                                                                                                                                                                                                                                         |
| 9-10  | ?      | ?          | ?                                                                                                                                                                                                                                              |
| 11-12 | Uint16 | Little     | Weight<br> <br>- For lbs divide by 100<br> - For kg divide by 200<br>                                                                                                                                                                          |

The line from the wiki page, "But be aware that the weight date and time is only valid if the stabilized flag is true and the weight removed flag is false!" still rings true.

I thought maybe you could update the wiki with this info, so it can help others like the info that's already there did for me :)

oliexdev, 2020-04-08:
what do you want to do? The mi scale v2 is already supported and what you found out is written in the wiki?

BeMacized, 2020-04-08:
From what I can see, only the format of the advertisement data of the v1 version is described on the wiki. The format differs for the v2 version. I'm asking for the v2 format description to be added to the wiki as well.

oliexdev, 2020-04-08:
The only difference is that the mi scale v2 sends the impedance value. Please be aware that you have to comply to the GPLv3 if you use openScale source-code, so any derivative work have to be the GPLv3 license.

closed 2020-04-08 by oliexdev

BeMacized, 2020-04-08:
I'm getting the feeling you're not fully understanding where I'm coming from. I've opened this issue in regards to the repository _wiki_, not the openScale source code. Please take a look at the table in the original issue and the first table on the wiki page I linked.

The v2 version's advertising data indeed contains the same information, but it differs from the v1 model. For example, there are two status bytes rather than one, of which the bits are in a different order. The weight bytes appear at the end of the advertising data rather than straight after the status bytes.

Surely that information could be useful to someone were it to be added to the wiki?