oliexdev/openScale #566: Scale support report: Quanta QTBL10

rmansilha, 2020-04-05:
Hi, 

I'm glad to share with you and your community that openScale works with Quanta QTBL10 (http://www.quantaproducts.co/produto/QTBL10/balan.html). The weight and BMI metrics were successfully checked against values obtained from KeepFit (https://play.google.com/store/apps/details?id=com.beikang), the app recommended by Quanta. Below is a picture of my scale:
[image]

I found your project through a closed issue (https://github.com/oliexdev/openScale/issues/140) while googling for an alternative to KeepFit. I'm not using KeepFit mostly due to privacy concerns. For example, it requires permission for making phone calls.

Thus, I suggest to add Quanta QTBL10 to the list of supported scales and, perhaps, a create "Scale support report" kind of issue. 

Please, let me know if you need further information.  

Congrats for the very good work and thanks for your attention,
Rodrigo.

oliexdev, 2020-04-08:
Thanks, could you tell me what is shown when you found the device in openScale? Which device driver is used?

rmansilha, 2020-04-08:
Hi, the device is "YunChen" (as in issue #140).

oliexdev, 2020-04-09:
Thanks, I added it to the wiki

closed 2020-04-09 by oliexdev