oliexdev/openScale #567: OpenScale Sync MQTT send ALL data

ivanfor, 2020-04-08:
Hi.
I've noticed that MQTT sync of OpenScale Sync only sends weight data.
I would expect to receive everything (fat, water, muscle, etc).
Would it be possible?

Moreover: why is it using "a" and "b" tags instead "date" and "weight"??
```
openScaleSync/measurements/insert {"a":1586322686000,"b":81.85}
openScaleSync/measurements/insert {"a":1586236281000,"b":81.5}
openScaleSync/measurements/insert {"a":1586150050000,"b":80.95}
openScaleSync/measurements/insert {"a":1586000585000,"b":80.7}
openScaleSync/measurements/insert {"a":1585897365000,"b":81.35}
openScaleSync/measurements/insert {"a":1585851795000,"b":80.0}
openScaleSync/measurements/insert {"a":1585774169000,"b":81.1}
openScaleSync/measurements/insert {"a":1585717868000,"b":81.55}
openScaleSync/measurements/insert {"a":1585717739000,"b":83.1}
openScaleSync/measurements/insert {"a":1585687430000,"b":82.0}
openScaleSync/measurements/insert {"a":1585687382000,"b":82.0}

```
Thanks!!!

oliexdev, 2020-04-09:
Currently only weight is supported, see the wiki.

It should be date and weight, could you attach the debug log file of both apps.

ivanfor, 2020-04-09:
Hi. 
How can I get the debug log? Is the development version of the app needed?

On the other hand, I know google fit only supports weight, but my commentary of getting muscle, wather, fat, etc data in MQTT was kind of an enhancement proposal. ¿Would It be possible?

Thanks.

oliexdev, 2020-04-09:
On the start openScale sync screen there is a switch where you can enable the debug log.
And no currently there is no plan to sync more measurements.

ivanfor, 2020-04-11:
> On the start openScale sync screen there is a switch where you can enable the debug  #log

Not sure it this option is working correctly: it seems to get switched off as long as I change from 'overview' to any other tab....
However, this is what I've gotten after a few attempts:

2020-04-11 08:29:11.304 Debug [2] c: Debug log enabled, openScale sync v0.3.1 (5), SDK 28, samsung SM-N960F
2020-04-11 08:29:13.389 Debug [2] a: Check GoogleFit sync status
2020-04-11 08:29:13.390 Error [2] a: Google sign in expired
2020-04-11 08:29:13.663 Error [2] a$a: GoogleFit sign in failed
2020-04-11 08:29:14.563 Debug [2] View: Google sign in button clicked
2020-04-11 08:29:15.454 Debug [2] a: GoogleFit permission granted
2020-04-11 08:29:18.819 Debug [2] c: openScale version 48 with content provider API version 1
2020-04-11 08:29:18.821 Debug [2] c: openScale version is ok(com.health.openscale)
2020-04-11 08:29:18.831 Debug [2] c: openScale user found(Ivan)
2020-04-11 08:29:18.831 Debug [2] c: Check GoogleFit sync status
2020-04-11 08:29:18.832 Debug [2] a: GoogleFit successful sign in
2020-04-11 08:29:18.833 Debug [2] c: Check MQTT sync status
2020-04-11 08:29:18.833 Debug [2] c: Trying to connect to MQTT server
2020-04-11 08:29:18.886 Debug [2] c: openScale version 48 with content provider API version 1
2020-04-11 08:29:18.888 Debug [2] c: openScale version is ok(com.health.openscale)
2020-04-11 08:29:18.893 Debug [2] d$b: Successful connected to tcp://mqtt.casa
2020-04-11 08:29:18.893 Debug [2] d$b: Successful connected to tcp://mqtt.casa
2020-04-11 08:29:18.895 Debug [2] d$b: Successful connected to tcp://mqtt.casa
2020-04-11 08:39:17.167 Error [2] d$b: Failed to connect to tcp://mqtt.casa Conexión ya en curso (32110)
2020-04-11 08:39:17.170 Error [2] d$b: Failed to connect to tcp://mqtt.casa Conexión ya en curso (32110)
2020-04-11 08:39:17.188 Error [2] d$b: Failed to connect to tcp://mqtt.casa Conexión ya en curso (32110)

ivanfor, 2020-04-11:
> And no currently there is no plan to sync more measurements.

Would you consider opensourcing openscale sync and accepting contributions?

oliexdev, 2020-04-20:
The problem with the "a" and "b" tags in MQTT should be fixed with the next release. 

> Would you consider opensourcing openscale sync and accepting contributions?

No not at the moment.

closed 2020-04-20 by oliexdev