oliexdev/openScale #568: (re-open) openScale sync regularly stops syncing to GoogleFit (error "GoogleFit sync failed")

PavelGorbanj, 2020-04-08:
This is a re-open of #546, as at least one more person (+ @ivanfor) experiences the same bug.

After a day, openScale Sync stops syncing to GoogleFit (error "GoogleFit sync failed").
It reproduces on 4 devices for 2 users.

Steps to reproduce the behavior:
1. Configure synchronisation to Google Fit
2. Weight, check that new measure appeared in Google Fit
3. Wait for a day
4. Weight, open Sync and get error "GoogleFit sync failed" (see screenshots).
5. Open configuration screen in Sync, fix issue by clicking "Google sign in", then "Manual full sync"
6. Weight, check that new measure appeared in Google Fit
3. Wait for a day, open Sync and get error "GoogleFit sync failed" (see screenshots).
repeat:)

Reproduced with Sync version 0.3.1, Pro version 2.2.0-pro

**Expected behavior**
Sync doesn't require manual intervention.

**Additional context**
I don't know if it matters, but there are several users configured on the scale (and corresponding ones are chosen in Sync).
The bug is reproduced on 4 different Samsung devices (Android 8, Android 10) with 2 different accounts.
Both openScale Pro and Sync were bought/installed from Google Play.

**Debug log**
... in progress ...
[image] [image]

PavelGorbanj, 2020-04-08:
Just in case, This bug happens every day. No other app with google sign-in behaves like that on my phone (and I use such sign-in method a lot) and also another person beside @ivanfor.
Also, I do not disconnect my account.

ivanfor, 2020-04-09:
Hi. I have the same behaviour than @PavelGorbanj
It happens very frecuently. Probably more than once a day. I don't have any other google account on the phone either.
Is there anything we can do to help you debugging?

oliexdev, 2020-04-09:
please don't create a duplicate issue.

closed 2020-04-09 by oliexdev