oliexdev/openScale #570: Fetching measurements from Sanitas SBF70 suddenly throws an error

cricktor, 2020-04-15:
**Describe the bug**
I do step on my Sanitas SBF70 here an then accumulate some measurements before I fire up openScale pro to fetch the recent ones. Last time it worked flawlessly was on March, 27th 2020.
Today I wanted to fetch the past measures (maybe 2 to 4, not sure). My settings are that the scale is not connected automatically when I open the app. I tap manually on the Bluetooth icon and usually the scale is seen, connected and recent data is fetched without issues, but not so today.

**To Reproduce**
Steps to reproduce the behavior:
1. Tap on Bluetooth icon to connect to scale
2. As soon as the bubble "Connection established" (I run my version in German language where it says "Verbindung hergestellt") shows up the app crashes and the helpless bug screen is shown
3. This can be reproduced everytime now (a restart of the phone or turning off Bluetooth doesn't change the behavior)
4. I haven't had this problem before

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): no

**Expected behavior**
Normal fetching of stored past measurements as it worked flawlessly before.

**Additional context**
Error details captured from bug screen:
Build version: 2.2.0-pro 
Current date: 2020-04-15 16:59:58 
Device: Samsung SM-N910F (on LineageOS, Android v9)
 
Stack trace:  
java.util.UnknownFormatConversionException: Conversion = ')'
	at java.util.Formatter$FormatSpecifier.conversion(Formatter.java:2781)
	at java.util.Formatter$FormatSpecifier.<init>(Formatter.java:2811)
	at java.util.Formatter$FormatSpecifierParser.<init>(Formatter.java:2624)
	at java.util.Formatter.parse(Formatter.java:2557)
	at java.util.Formatter.format(Formatter.java:2504)
	at java.util.Formatter.format(Formatter.java:2458)
	at java.lang.String.format(String.java:2842)
	at com.health.openscale.gui.MainActivity$10.handleMessage(MainActivity.java:568)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6724)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:495)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.
Because the app crashes to the bug screen I don't see much meaningful data in the saved debug log from above procedure. Attached a copy here anyway but removed personal data from it.
[openScale_2020-04-15_17-06.txt](https://github.com/oliexdev/openScale/files/4482193/openScale_2020-04-15_17-06.txt)

oliexdev, 2020-04-16:
Could you try to change the app language in the settings into a different language and test it again.

cricktor, 2020-04-16:
Language was on "System default", i.e. German for me.
Setting it to German explicitly gives again the "helpless bug" crash screen.
Setting it to English worked OK: four past stored measurements were fetched without an issue or error.
[But something must have changed between March 27th and midth of April, because it worked before with Language set to "System default" eq. "German".

cricktor, 2020-04-17:
One thing I haven't seen before is an additional bubble info after the bubble with "Connected" due to low battery state: I haven't captured the full message as it goes away rather quickly.
In parenthesis is a battery value shown as "...(3\%)...", looks a little bit incorrectly escaped.

oliexdev, 2020-04-18:
thanks the escape of the percentage symbol for the translation in German and other languages were not correct if the message "on low battery" was shown. Should be fixed with now.

closed 2020-04-18 by oliexdev