oliexdev/openScale #573: Import backup into mysql

dproetel, 2020-04-29:
Good morning,
I try to import the sqlite db into mysql to have all weights in a central database.
My challenge is the datetime. Where can I find how to convert on shell/mysql level the datetime int in something mysql understands?
Thanks in advance
Daniel

oliexdev, 2020-05-02:
What is your goal? The datetime is a long value which can be saved in mysql.

dproetel, 2020-05-02:
Hi,
My goal is to save it in the database and easily use it with mysql functions. Actually I cut off the last 3 digits and have a unix time stamp. It works, but is that right?

Ciao Daniel

oliexdev, 2020-05-02:
yes you have to interpret the long value depending on your goal. To which app to you want to connect your mysql database?

closed 2020-05-02 by oliexdev