oliexdev/openScale #574: RENPHO Bluetooth Body Fat Scale supported but not listed

teeks99, 2020-05-08:
**Scale name**
The RENPHO ES-CS20M scale was not listed on the supported scales page, but worked perfectly out of the box. 

It was detected by bluetooth as a "QN-Scale".

[image]
[image]
[image]

Successful log:
[openScale_2020-05-06_06-08.txt](https://github.com/oliexdev/openScale/files/4596557/openScale_2020-05-06_06-08.txt)

oliexdev, 2020-05-08:
thanks I added the scale into the wiki.

closed 2020-05-08 by oliexdev

giupardeb, 2020-11-17:
@teeks99 Can you explain the steps to add this scale on the app? openScale doesn't find it

teeks99, 2020-11-17:
I don't have this model anymore. However, there wasn't anything special. Open the app, pair with the scale when you stand on it.

giupardeb, 2020-11-17:
it doesn't work for me xD

guix77, 2021-09-02:
See #769