oliexdev/openScale #576: Error when changing user for any entry

cyannade, 2020-05-09:
I could change a couple of them to the correct user, but now I can't do it anymore (with any entry or user).

Build version: 2.2.0 
Current date: 2020-05-09 14:00:37 
Device: Xiaomi Redmi Note 5 
 
Stack trace:  
android.database.sqlite.SQLiteConstraintException: UNIQUE constraint failed: scaleMeasurements.userId, scaleMeasurements.datetime (code 2067 SQLITE_CONSTRAINT_UNIQUE)
	at android.database.sqlite.SQLiteConnection.nativeExecuteForChangedRowCount(Native Method)
	at android.database.sqlite.SQLiteConnection.executeForChangedRowCount(Unknown Source:31)
	at android.database.sqlite.SQLiteSession.executeForChangedRowCount(Unknown Source:15)
	at android.database.sqlite.SQLiteStatement.executeUpdateDelete(Unknown Source:20)
	at androidx.sqlite.db.framework.FrameworkSQLiteStatement.executeUpdateDelete(FrameworkSQLiteStatement.java:46)
	at androidx.room.EntityDeletionOrUpdateAdapter.handle(EntityDeletionOrUpdateAdapter.java:70)
	at com.health.openscale.core.database.ScaleMeasurementDAO_Impl.update(ScaleMeasurementDAO_Impl.java:171)
	at com.health.openscale.core.OpenScale.updateScaleData(OpenScale.java:382)
	at com.health.openscale.gui.activities.DataEntryActivity.saveScaleData(DataEntryActivity.java:395)
	at com.health.openscale.gui.activities.DataEntryActivity.onOptionsItemSelected(DataEntryActivity.java:218)
	at android.app.Activity.onMenuItemSelected(Unknown Source:68)
	at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:384)
	at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:219)
	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
	at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:64)
	at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:207)
	at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:840)
	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:991)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:981)
	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:151)
	at android.view.View.performClick(Unknown Source:17)
	at android.view.View.performClickInternal(Unknown Source:3)
	at android.view.View.access$3100(Unknown Source:0)
	at android.view.View$PerformClick.run(Unknown Source:2)
	at android.os.Handler.handleCallback(Unknown Source:2)
	at android.os.Handler.dispatchMessage(Unknown Source:4)
	at android.os.Looper.loop(Unknown Source:207)
	at android.app.ActivityThread.main(Unknown Source:107)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(Unknown Source:11)
	at com.android.internal.os.ZygoteInit.main(Unknown Source:274)

oliexdev, 2020-05-10:
Could you enable the openScale debug log and attach the result here when you change the user.

cyannade, 2020-05-10:
The captured log only shows the personal data I was trying to move.

However, I realised that the error showed when trying to move an entry to an user that already had that entry. I synced the history when I first connected the Mi Scale 2 (all of it on one user), then created more users and turned on Smart user assignment. The app relogged the entries to the specific users, but still showed them on my original user.

So it was an error caused by the duplicates. Maybe it could show a message when it detects an entry that already exists for another user?

oliexdev, 2020-05-15:
This shouldn't happened normally. Currently, I can't reproduce it but I added a conflict strategy to ignore this state.

closed 2020-05-15 by oliexdev