oliexdev/openScale #577: App Crash while sync

proebstle, 2020-05-26:
**Describe the bug**
Since today the app crash while sync over Bluetooth.

**Debug log**
Build version: 2.2.0-pro 
Current date: 2020-05-26 07:06:06 
Device: Blackview BV5900 
 
Stack trace:  
java.util.UnknownFormatConversionException: Conversion = ')'
	at java.util.Formatter$FormatSpecifier.conversion(Formatter.java:2781)
	at java.util.Formatter$FormatSpecifier.<init>(Formatter.java:2811)
	at java.util.Formatter$FormatSpecifierParser.<init>(Formatter.java:2624)
	at java.util.Formatter.parse(Formatter.java:2557)
	at java.util.Formatter.format(Formatter.java:2504)
	at java.util.Formatter.format(Formatter.java:2458)
	at java.lang.String.format(String.java:2842)
	at com.health.openscale.gui.MainActivity$10.handleMessage(MainActivity.java:568)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6702)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)

oliexdev, 2020-05-26:
see issue #570

closed 2020-05-26 by oliexdev

proebstle, 2020-05-26:
Thx .  Works eine now!
--
Diese Nachricht wurde von meinem Android Mobiltelefon mit 1&1 Mail gesendet.