oliexdev/openScale #578: Baby/Pet weighing

repomaa, 2020-05-26:
Weighing babies or pets is only possible by holding them in your arms while stepping on the scale yourself.

It'd be cool to add a new type of user to the app and automatically subtract the weight of another user from it when weighing. Body fat and other measurements wouldn't make sense here of course. The app could prompt the person to weigh themselves before weighing the baby or pet if the last recorded weight of the person is older than X hours.

repomaa, 2020-05-26:
I guess this is related to #559

oliexdev, 2020-05-26:
good idea and yes it is similar to issue #559 but it seems that @ofeskr had no interest in this feature anymore.

How would you name this mode in the user settings?

repomaa, 2020-05-30:
Maybe “Assisted weighing” would be generic enough? As in a person has to assist another person/animal with the weighing by holding them? Could be misunderstood though.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

oliexdev, 2020-06-17:
I implemented an assisted weighing as you proposed please try it out the latest [dev build](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and give feedback.

closed 2020-07-05 by oliexdev

repomaa, 2020-07-10:
Hey! Awesome! Thanks! I looked at the dev build, but can't find it anywhere in the interface. Do some preconditions have to be met?

oliexdev, 2020-07-19:
You have to check the assisted weighting in the user settings.

repomaa, 2020-07-22:
Hmm... Maybe I'm missing something.

[image]

oliexdev, 2020-07-26:
you didn't install the latest dev version, read again https://github.com/oliexdev/openScale/issues/578#issuecomment-645405628

repomaa, 2020-07-27:
i did... maybe because both, openScale and openScale dev are installed on the same device?

repomaa, 2020-07-27:
I opened the dev version though

oliexdev, 2020-07-27:
could you attach the openScale debug log file here?

gudvinr, 2021-01-10:
@oliexdev how does it work though? I added new user, enabled "Assisted weighing" in settings for new user and tried to stand on scales without and then with additional weight. It just append raw weight from second measurement

oliexdev, 2021-01-11:
@gudvinr should hopefully self explain. Just add a "normal" user and an "assisted weighing" user. Then weight the normal user first to have a reference weight. Then switch to the pet/weight user. Then weight again, but with your baby/pet in your arm. You should see a popup where you can select the reference user. That's all.
Always the last measurement of the reference user are subtracted.