oliexdev/openScale #581: Body composition for Custom Bluetooth Scale

ygreq, 2020-05-29:
I have a broken Withings / Nokia Body + Smart Scale that does measure Body composition. It would be great if I can also use those functions in a custom scale built based on this tutorial: https://github.com/oliexdev/openScale/wiki/Custom-Bluetooth-Scale

Any ideas if this has been done already? Or any suggestions on how it can be done?

Thank you so much!

oliexdev, 2020-06-01:
Well I don't think that have somebody done it but how it's done it is detailed describes in the tutorial? So what is your specific question?

ygreq, 2020-06-01:
Hi oliexdev! I don't understand your first question. And related to your second question, do you think it is feasible to do a diy Body Composition smart scale? Any pointers on what is needed?

Thank you!

oliexdev, 2020-06-02:
I am sorry, I still don't get your goal?
You have a broken smart scale? What is broken?
What do you want to do with those broken smart scales?
Do you want to build a complete smart scale from scratch using electronic parts?
Do you want to repair your broken smart scales?

closed 2020-06-10 by oliexdev