oliexdev/openScale #585: Change default goal from arbitrary weight to top of healthy body fat percentage range

ethana2, 2020-06-10:
When I installed this application from F-Droid, I was asked to set a goal weight. But neither weight nor BMI is directly related to health-- Body composition is.

When asked for an initial goal, it should instead take the form of body fat percentage, and it should populate by default with the top end of the healthy body fat percentage range for the user's age and sex.

oliexdev, 2020-06-10:
interesting idea but what we do if somebody want to gain the body fat percentage or don't have a scale that supports body fat?

ethana2, 2020-06-10:
Instead of asking users for their current weight when they install the application, they could be asked, first, to perform an initial sync with their scale. This would determine their current body weight and/or composition as well as scale capabilities. They could then be asked for their goals in a way that takes this information into consideration.

Users that are determined to be below healthy body fat such as those fighting anorexia would have their goals default populated with the middle of the healthy body fat percentage range if their scale supports body composition, or with the weight that is most likely to correspond to it given sex, age, and height.

oliexdev, 2020-06-12:
What if you do with users that doesn't have a Bluetooth scale and have to manually enter their weight? All in all a good idea but very complex to implement.

ethana2, 2020-06-12:
I was thinking a scale sync page with three options: Bluetooth, barcode, manual-only. If they select manual only then we assume whatever they enter is what their scale provides

oliexdev, 2020-06-12:
would be an option but the barcode method is really exotic and could be only useful with your scale. Maybe we moved this barcode method as an extra point only for your scale. Other users might be irritated of this method.

ethana2, 2020-07-01:
Yeah, I'll go ahead and defer to you on how you want this appsync fragment tied in once I get that far.

lbdroid, 2021-06-14:
What about a few more details in the goal setting script? Instead of just asking the use what their weight goal is, ask them if they want to set a goal, then ask them what metric to use, then suggest a value as a starting point with a spinner to adjust the goal per the user's preference.

closed 2023-02-12 by oliexdev