oliexdev/openScale #586: Navigation refactor

ethana2, 2020-06-12:
Currently openScale uses a mixture of activities, fragments, and "slides"-- fragments managed through  the AppIntro library. What this means is that when a user navigates to Settings or Bluetooth Settings, they're navigating to a different activity and not just a different fragment. In doing so, they lose all on-screen navigation interface and can only navigate back by using the back button. This makes the UI feel discordant, and there's no benefit because the resulting screen real estate is wasted with empty space.

I suggest that BaseAppCompatActivity be merged into MainActivity and all other activities be converted into fragments, and then for it to be made into a NavHost and for a NavHostController to be created inside of it and used to manage all application navigation.

Additionally, upon first application launch, the user is showered with slides, many of which contain very little information and much of that information only of interest to license- or privacy-focused people. This information should be in the Google Play and F-Droid pages, and in an About fragment or two.

Only two fragments should be shown to users upon first launch:
1. Sync scale
2. Manually provide user info (height, sex, etc)
For this, the use of AppIntro is overkill; this should be handled through normal navigation and this dependency removed.

oliexdev, 2020-06-12:
> Currently openScale uses a mixture of activities, fragments, and "slides"-- fragments managed through the AppIntro library. What this means is that when a user navigates to Settings or Bluetooth [...]

Already done it in the Androidx branch but the MeasurementSettings have to be still refactored because the ordering of the measurements doesn't work good. Please test it.

> Additionally, upon first application launch, the user is showered with slides, many of which contain very little information and much of that information only of interest to license- or [...]

Trust me only few people will read the Google Play or F-Droid pages. I introduced the slide show that the highlights are shown and users are tend to read the slide more often.

> Only two fragments should be shown to users upon first launch:
> 
>     Sync scale
>     Manually provide user info (height, sex, etc)
>     For this, the use of AppIntro is overkill; this should be handled through normal navigation and this dependency removed.

That was the previous approach and you can still skip the complete intro any time. Then the user menu is only shown.

ethana2, 2020-06-12:
> Already done it in the Androidx branch but the MeasurementSettings have to be still refactored because the ordering of the measurements doesn't work good. Please test it.

Ope, sorry, forgot to check that. I'll try to be more thorough going forward.
Regarding the rest, noted

ethana2, 2020-06-12:
I played around with the MeasurementPreferences a bit.. When say the ordering of the measurements doesn't work good, do you mean that having it all thrown into one list without grouping is unwieldy from a ux perspective, or something else?

This is my first time diving into an app like this-- I'm actually having trouble *finding* the layout xml for MeasurementPreferences. I see MeasurementPreferences.java where it extends androidx.preference.PreferenceFragmentCompat, and I see the nav_measurement_preferences in the mobile navigation, and in res/layout I see measurement_preferences_widget_layout.xml which is the layout for the individual items... but I can't for the life of me find the layout.xml for the MeasurementPreferences fragment. Could you please point me in the right direction on that?

oliexdev, 2020-06-12:
> I played around with the MeasurementPreferences a bit.. When say the ordering of the measurements doesn't work good, do you mean that having it all thrown into one list without grouping is unwieldy from a ux perspective, or something else?

No sometimes if you scroll down and then drag and drop a measurement it will append in the end of the list not at the drag position. Second problem is that the on/off switch are not saved if you scroll the measurement. All in all I wanted to replace the list with a recycler view with an adapter class were you can more easily use the internal drag and drop system.

> This is my first time diving into an app like this-- I'm actually having trouble finding the layout xml for MeasurementPreferences. I see MeasurementPreferences.java where it extends [...]

You can find the xml file under res/xml/measurement_preferences.xml

closed 2020-06-16 by oliexdev