oliexdev/openScale #592: Disable tabs?

woj-tek, 2020-07-06:
**Is your feature request related to a problem? Please describe.**
Since version 2.3.0, when hitting back on the "Table" view I'm taken to main screen instead of leaving the app (as before). The small issue is that drawing graphs on main screen is a little bit slow so the app is getting stuck for 1-2s with mixed view of the table and main screen graph and only then allows me to close it.

**Describe the solution you'd like**
My use-case is quite simple - note daily weight, so I basically use only table view. Either bringing back previous workflow (back = exit app) or allowing to disable unused tabs (so back would effectively close the app) would work

**Describe alternatives you've considered**
Alternatively I can just use `home` button, but this a little bit inconsistent and virtually all other app requires you to exit it using "back-back-back' or else it will stay on the last activity.

oliexdev, 2020-07-06:
thanks should be fixed.

closed 2020-07-06 by oliexdev

woj-tek, 2020-07-06:
Awesome, thanks!