oliexdev/openScale #594: Unable to Connect to ES-26M Scale

ZachMyers3, 2020-07-07:
**Describe the bug**
The ES-26M is on the list of supported devices but for the life of me I'm unable to connect to it. I'm able to sync with the FITINDEX official application, but the device does not appear when searching for the smart scale.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to settings
2. Click on Bluetooth
3. It will search for the device and not find a result
4. See error

**Expected behavior**
The smart scale to be detected when in use, the same time the sync occurs on the official app.

**Debug log**
```
2020-07-07 17:58:06.759 Debug [2] AboutPreferences: Debug log enabled, openScale v2.3.0 (49), SDK 29, asus ASUS_I01WD
2020-07-07 17:58:06.763 Debug [2] AboutPreferences: Selected user id(1) name(Zach Myers) birthday(Sat Apr 02 00:00:00 EST 1994) age(26) body height(185.42) scale unit(lb) gender(male) initial weight(83.37) goal weight(77.11) goal date(Tue Jan 05 00:00:00 EST 2021) measure unt(in) activity level(1)
2020-07-07 17:58:10.069 Info [2] BluetoothCentral: scan started
2020-07-07 17:58:10.492 Debug [2] BluetoothSettingsFragment: Found unsupported device Hue Lamp [C0:80:F1:CC:D7:6E]
2020-07-07 17:58:11.564 Debug [2] BluetoothSettingsFragment: Found unsupported device Hue Lamp [D1:CD:D2:58:2C:A7]
2020-07-07 17:58:23.745 Info [2] BluetoothCentral: scan stopped
2020-07-07 17:58:30.321 Info [2] BluetoothCentral: scan started
2020-07-07 17:58:30.421 Debug [2] BluetoothSettingsFragment: Found unsupported device Hue Lamp [D1:CD:D2:58:2C:A7]
2020-07-07 17:58:31.066 Debug [2] BluetoothSettingsFragment: Found unsupported device Hue Lamp [C0:80:F1:CC:D7:6E]
2020-07-07 17:58:32.566 Info [2] BluetoothCentral: scan stopped
2020-07-07 17:58:37.175 Info [2] BluetoothCentral: scan started
2020-07-07 17:58:37.932 Debug [2] BluetoothSettingsFragment: Found unsupported device Hue Lamp [C0:80:F1:CC:D7:6E]
2020-07-07 17:58:38.146 Debug [2] BluetoothSettingsFragment: Found unsupported device Hue Lamp [D1:CD:D2:58:2C:A7]
2020-07-07 17:58:57.182 Info [2] BluetoothCentral: scan stopped
2020-07-07 17:58:59.394 Info [2] BluetoothCentral: no scan to stop because no scan is running
2020-07-07 17:59:02.345 Debug [2] AboutPreferences: Debug log disabled
```

oliexdev, 2020-07-08:
Did you pair the scale with the original app? Could you try to unpair the Bluetotth device or could you try to uninstall the original app and reset your scale (do a backup of your data before!) and try again with openScale.

ZachMyers3, 2020-07-08:
> Did you pair the scale with the original app?

Yes, and this works as expected.

> Could you try to unpair the Bluetotth device or could you try to uninstall the original app and reset your scale (do a backup of your data before!) and try again with openScale.

I'll look into this option and get back with results. 

When I initially purchased the scale I attempted to set the scale up without the original app (exclusively with openScale) and it was still having the same problem, so then I installed the original app and paired to see if that would work. I'll try this again after resetting my scale and uninstalling the original app.

Thanks for you help!

closed 2020-07-19 by oliexdev