oliexdev/openScale #599: Sanitas SBF70

FreddyPL, 2020-07-15:
Hallo,
I have Sanitas SBF70 with BT scale.
Somethink was wrong :( and I must recalibrate scale.
It display 75kg instead of 100kg - is it possible?
Did You know something about new firmware, or how to recalibrate scale

Regards

FreddyPL

oliexdev, 2020-07-19:
no I don't know about a new firmware please attach the openScale debug log here

FreddyPL, 2020-07-19:
Here You are

> 2020-07-19 13:36:36.946 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.1.2-dev_49fc3aca_2019-10-20 (1571577752), SDK 29, samsung SM-A505FN
2020-07-19 13:36:36.947 Debug [2] AboutPreferences: Selected user id(1) name(Janusz) birthday(Wed Aug 06 00:00:00 GMT+02:00 1958) age(61) body height(182,00) scale unit(kg) gender(male) initial weight(92,00) goal weight(82,00) goal date(Tue Jan 19 12:36:16 GMT+01:00 2021) measure unt(cm) activity level(0)
2020-07-19 13:37:43.089 Debug [2] OpenScale: Disconnecting from bluetooth device
2020-07-19 13:37:43.093 Debug [2] BluetoothCommunication: Bluetooth disconnect
2020-07-19 13:37:43.095 Info [2] BluetoothCentral: no scan to stop because no scan is running
2020-07-19 13:37:43.098 Error [2] BluetoothCentral: cannot cancel connection to unknown peripheral D0:B5:C2:CC:3F:37
2020-07-19 13:37:44.745 Debug [2] OpenScale: Trying to connect to bluetooth device [D0:B5:C2:CC:3F:37] (SANITAS SBF70)
2020-07-19 13:37:44.748 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2020-07-19 13:37:44.766 Info [2] BluetoothCentral: scan started
2020-07-19 13:37:44.767 Debug [2] BluetoothCommunication: Stop machine state
2020-07-19 13:37:47.750 Debug [2] BluetoothCommunication: Found peripheral 'SANITAS SBF70'
2020-07-19 13:37:47.768 Info [2] BluetoothCentral: scan stopped
2020-07-19 13:37:48.773 Debug [2] BluetoothCommunication: Try to connect to BLE device D0:B5:C2:CC:3F:37
2020-07-19 13:37:48.909 Info [2] BluetoothPeripheral: connect to 'SANITAS SBF70' (D0:B5:C2:CC:3F:37) using TRANSPORT_LE
2020-07-19 13:37:51.400 Info [99005] BluetoothPeripheral: connected to 'SANITAS SBF70' (BOND_NONE) in 2,4s
2020-07-19 13:37:51.410 Debug [2] BluetoothPeripheral: discovering services of 'SANITAS SBF70' with delay of 0 ms
2020-07-19 13:37:52.281 Info [99005] BluetoothPeripheral: discovered 5 services for 'SANITAS SBF70'
2020-07-19 13:37:52.298 Debug [2] BluetoothCommunication: connected to 'SANITAS SBF70'
2020-07-19 13:37:52.300 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2020-07-19 13:37:52.302 Debug [2] BluetoothCommunication: Resume machine state
2020-07-19 13:37:52.305 Debug [2] BluetoothCommunication: Step Nr 0
2020-07-19 13:37:52.307 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2020-07-19 13:37:52.310 Debug [2] BluetoothCommunication: Stop machine state
2020-07-19 13:37:52.345 Debug [2] MainActivity: Bluetooth connection successful established
2020-07-19 13:37:52.437 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2020-07-19 13:37:52.440 Debug [2] BluetoothCommunication: Resume machine state
2020-07-19 13:37:52.442 Debug [2] BluetoothCommunication: Step Nr 1
2020-07-19 13:37:52.446 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2020-07-19 13:37:52.449 Debug [2] BluetoothCommunication: Stop machine state
2020-07-19 13:37:52.457 Debug [2] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-07-19 13:37:52.518 Debug [2] BluetoothCommunication: SUCCESS: Writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-07-19 13:37:52.521 Debug [2] BluetoothBeurerSanitas: Got init ack from scale; scale is ready
2020-07-19 13:37:52.523 Debug [2] BluetoothCommunication: Resume machine state
2020-07-19 13:37:52.526 Debug [2] BluetoothCommunication: Step Nr 2
2020-07-19 13:37:52.534 Debug [2] BluetoothCommunication: Invoke write bytes [E9 5F 14 30 90] on 0xffe1
2020-07-19 13:37:52.537 Debug [2] BluetoothCommunication: Step Nr 3
2020-07-19 13:37:52.545 Debug [2] BluetoothCommunication: Invoke write bytes [E7 4F 00 00 00 00 00 00 00 00] on 0xffe1
2020-07-19 13:37:52.547 Debug [2] BluetoothCommunication: Stop machine state
2020-07-19 13:37:52.552 Debug [2] BluetoothPeripheral: writing <e95f143090> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-07-19 13:37:52.609 Debug [2] BluetoothCommunication: SUCCESS: Writing <E9 5F 14 30 90> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-07-19 13:37:52.617 Debug [2] BluetoothPeripheral: writing <e74f0000000000000000> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-07-19 13:37:52.621 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 66,40
2020-07-19 13:37:52.702 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 4F 00 00 00 00 00 00 00 00> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-07-19 13:37:52.743 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 65,90
2020-07-19 13:37:53.778 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 67,05
2020-07-19 13:37:54.948 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 68,10
2020-07-19 13:37:56.077 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 69,80
2020-07-19 13:37:57.214 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 68,90
2020-07-19 13:37:58.315 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 68,75
2020-07-19 13:37:59.464 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 67,70
2020-07-19 13:38:00.613 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 66,55
2020-07-19 13:38:01.716 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 66,95
2020-07-19 13:38:02.864 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 63,40
2020-07-19 13:38:04.064 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 0,00
2020-07-19 13:38:09.637 Debug [2] AboutPreferences: Debug log disabled

closed 2020-07-19 by FreddyPL