oliexdev/openScale #603: Make openScale resizable

alexanderadam, 2020-07-23:
**Is your feature request related to a problem? Please describe.**

I'm using [Samsung DeX](https://www.samsung.com/us/explore/dex/), which gives a desktop like experience and makes use of Android windowing support.
openScale works fine on that except that it isn't resizable.

**Describe the solution you'd like**

Regarding [Samsung documentation this on how to enable window resizing for Android applications](https://developer.samsung.com/samsung-dex/modify-optimizing.html):

```xml
<application
    android:resizeableActivity="true">
</application>
```
This documentation also has more tweaks and information on how to improve the desktop experience in case you're interested.
They also have [an app testing guide](https://developer.samsung.com/samsung-dex/testing.html) (although some apps are fine with this guide but won't work on "real" DeX sometimes).

**Describe alternatives you've considered**

Not resizing openScale

**Additional context**

At the moment the resizing button is greyed out and the application is always nearly squarish:

[image]

**PS:** I know that it is possible to _force_ all apps to be resizable via `Samsung Labs` > `Force Resize` but it would obviously better to make the application natively resizable to create less friction.

oliexdev, 2020-07-26:
Didn't know this feature. It would be great if you could test this feature with openScale to see if it works well. It that possible?

alexanderadam, 2020-07-27:
Of course. If you have an APK I will happily test it. :+1:

oliexdev, 2020-07-27:
You can create your own APK using the source code and a little bit effort to test and implement your own feature request. Please send a PR if your are done. Thanks in advance.

alexanderadam, 2020-07-27:
I'm not an Android developer and I have no experience with building stuff for the Android ecosystem.
But I can try building it if you can recommend a build chain within a Docker container for me (or any other way how I can get a Apk without having to modify my system).
I would also use a VM in the worst case.

Or maybe you have any other idea?

oliexdev, 2020-07-27:
everybody has to start at some point. You only need Android Studio. You can find a guide at https://developer.android.com/guide
I never used Docker but I am sure you will find some information on Google.