oliexdev/openScale #607: Error opening window Add User

husniadil, 2020-08-10:
**Describe the bug**
An exception has occurred when trying to add a user for the first time.

**To Reproduce**
Steps to reproduce the behavior:
I use Samsung Galaxy S20. Open the app, on the wizard, I click next then it offers me to Add User, and I click the button, then the exception appears.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): Latest version when this issue is created

**Expected behavior**
It should handle the exception.

**Debug log**
```
Build version: 2.3.0 
Current date: 2020-08-10 08:17:24 
Device: Samsung SM-G980F 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.slides.SlideToNavigationAdapter}: java.lang.IllegalStateException: This Activity already has an action bar supplied by the window decor. Do not request Window.FEATURE_SUPPORT_ACTION_BAR and set windowActionBar to false in your theme to use a Toolbar instead.
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3555)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3707)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2220)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loop(Looper.java:237)
	at android.app.ActivityThread.main(ActivityThread.java:8016)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076)
Caused by: java.lang.IllegalStateException: This Activity already has an action bar supplied by the window decor. Do not request Window.FEATURE_SUPPORT_ACTION_BAR and set windowActionBar to false in your theme to use a Toolbar instead.
	at androidx.appcompat.app.AppCompatDelegateImpl.setSupportActionBar(AppCompatDelegateImpl.java:421)
	at androidx.appcompat.app.AppCompatActivity.setSupportActionBar(AppCompatActivity.java:150)
	at com.health.openscale.gui.slides.SlideToNavigationAdapter.onCreate(SlideToNavigationAdapter.java:43)
	at android.app.Activity.performCreate(Activity.java:7957)
	at android.app.Activity.performCreate(Activity.java:7946)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1307)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3530)
	... 11 more
```

lattice191011, 2020-08-14:
Same error as above with Andoid 6.0.1; until Openscale 6.1.0 it works fine!

Build version: 2.3.0 
Build date: 1980-01-01 00:00:00 
Current date: 2020-08-14 14:22:10 
Device: Samsung SM-G900F 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.slides.SlideToNavigationAdapter}: java.lang.IllegalStateException: This Activity already has an action bar supplied by the window decor. Do not request Window.FEATURE_SUPPORT_ACTION_BAR and set windowActionBar to false in your theme to use a Toolbar instead.
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3253)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3349)
	at android.app.ActivityThread.access$1100(ActivityThread.java:221)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1794)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:158)
	at android.app.ActivityThread.main(ActivityThread.java:7224)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
Caused by: java.lang.IllegalStateException: This Activity already has an action bar supplied by the window decor. Do not request Window.FEATURE_SUPPORT_ACTION_BAR and set windowActionBar to false in your theme to use a Toolbar instead.
	at androidx.appcompat.app.AppCompatDelegateImpl.setSupportActionBar(AppCompatDelegateImpl.java:421)
	at androidx.appcompat.app.AppCompatActivity.setSupportActionBar(AppCompatActivity.java:150)
	at com.health.openscale.gui.slides.SlideToNavigationAdapter.onCreate(SlideToNavigationAdapter.java:43)
	at android.app.Activity.performCreate(Activity.java:6876)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1135)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3206)
	... 9 more

oliexdev, 2020-08-14:
should be fixed in the latest dev version, see issue #595

closed 2020-08-15 by oliexdev