oliexdev/openScale #608: Error after pairing device

husniadil, 2020-08-10:
**Describe the bug**
An exception has occurred after pairing device.

**To Reproduce**
For the first time, open Settings, Bluetooth. Connect/Pair the app to Mi Scale 2. After pairing, press Back button.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 2.3.0

**Expected behavior**
It should handle the exception.

**Debug log**
```
Build version: 2.3.0 
Current date: 2020-08-10 08:28:10 
Device: Samsung SM-G980F 
 
Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{4331c89} (dedf956c-52bb-480e-8787-b8e6910778f0)} not attached to a context.
	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
	at androidx.fragment.app.Fragment.getResources(Fragment.java:869)
	at androidx.fragment.app.Fragment.getString(Fragment.java:891)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$2.run(BluetoothSettingsFragment.java:160)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:237)
	at android.app.ActivityThread.main(ActivityThread.java:8016)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076)

```

oliexdev, 2020-08-14:
Could you reproduce this error?

herrwusel, 2020-08-15:
I can confirm this behaviour on a HavocOS 3.7 with microg device:
```
Build version: 2.3.0 
Build date: 1979-11-30 01:00:00 
Current date: 2020-08-14 15:07:07 
Device: Xiaomi Redmi 8 
 
Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{b497dc5} (32a9c576-034c-49f4-b75a-52090669ae44)} not attached to a context.
	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
	at androidx.fragment.app.Fragment.getResources(Fragment.java:869)
	at androidx.fragment.app.Fragment.getString(Fragment.java:891)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$2.run(BluetoothSettingsFragment.java:160)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7356)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:915)
```

It is also happening after trying to get a measurement from the scale.

oliexdev, 2020-08-15:
should be fixed please try out the latest dev build

closed 2020-08-15 by oliexdev

herrwusel, 2020-08-16:
I tried the fix and the app isn't crashing. But after connecting successfully the app seems to disconnect right away and therefore isn't fetching any data (the bluetooth icon appears shortly and then disappears on my Yunmai). Don't know if it is related to this issue.

oliexdev, 2020-08-16:
please attach the openScale debug log file here

herrwusel, 2020-08-17:
here it is: [openscale.txt](https://github.com/oliexdev/openScale/files/5082919/openscale.txt)