oliexdev/openScale #611: Too dark icons/text in dark mode

naoliv, 2020-08-15:
**Describe the bug**
With the new dark theme some elements are difficult to read sometimes, specially when they are in black color (icons and text):

[image]
[image]
[image]
[image]
[image]

**Expected behavior**
Black icons/texts should probably be lighter (gray or white).

oliexdev, 2020-08-16:
which openScale version and which Android version do you using?

naoliv, 2020-08-16:
Oh, sorry.
It's openScale `2.3.1-light` with Android 10.

oliexdev, 2020-08-16:
Unfortunately, I can't reproduce this behaviour!? Did you enable the dark mode in the openScale settings? Did you enable the Night mode globally in Android?

oliexdev, 2020-08-16:
Do you have some special theme installed on your smartphone?

naoliv, 2020-08-16:
Yes, dark mode is enabled globally in Android.
In openScale it's saying that it's using the `Light` theme

[image]

naoliv, 2020-08-16:
I didn't change anything in openScale theme, but maybe it should follow the global Android setting? (regarding the dark mode)
By forcing `Dark` now in openScale the objects are properly changed to a light theme.

oliexdev, 2020-08-16:
Thanks now I can reproduce this error. For the moment you can switch it to the dark mode in openScale that would fix that problem but I need to fix this bug

oliexdev, 2020-08-16:
should be fixed now

closed 2020-08-16 by oliexdev