oliexdev/openScale #612: Needed Beta Tester for the trendline feature as Hacker's Diet (Libra like)

oliexdev, 2020-08-16:
I need some beta tester for the trendline feature as implemented in the [Hacker's Diet](https://www.fourmilab.ch/hackdiet/e4/pencilpaper.html)or Libra like trendline, see issue #436 .

Please install the [openScale-dev_Trendline-build.apk](https://github.com/oliexdev/openScale/releases/tag/travis-dev_Trendline-build) and give feedback if the calculation is correct and so on.

Marwel, 2020-08-17:
I imported my data from openscale into the dev version, so that i can look at real data.
The trendline looks strange to me.
The trendline goes up, when my actual measurement is going down...which is not, what i anticipated.
Also the numbers on that line do stack up a bit.
Afaiks in the code, you are using for the reference 1 day, am i right?
Even with that, the line should look different.
Also, why the multiply(0.1f)?

Screenshot
https://photos.app.goo.gl/xF2zkfHPC67z3d1g6

oliexdev, 2020-08-18:
thanks for your reply.

> The trendline looks strange to me.
> The trendline goes up, when my actual measurement is going down...which is not, what i anticipated.

I reversed the calculation, should be fixed in the latest build, please try it out.

>  Also the numbers on that line do stack up a bit.

Indeed, you can disable the labels in the settings.

>  Afaiks in the code, you are using for the reference 1 day, am i right?

I used the first measurement at the starting point.

> Also, why the multiply(0.1f)?

I used this formula `Tn=Tn−1+0.1(Wn−Tn−1)` from the Hackers Diet.

Marwel, 2020-08-18:
￼￼Looks better.
But i think the moving average formula is the real deal in this kind of situation.
The other formula was for temperatures.

https://www.fourmilab.ch/hackdiet/e4/pencilpaper.html#PencilMovingAverages
The specific formula should be 
https://www.fourmilab.ch/hackdiet/e4/figures/pencil6.png

It depends on the number of measurements, so that you can only get an trendline entry after that number of measurements.
quote:
"We can't compute a four day moving average until we have four days worth of data. That's why the first moving average in this example is A4."
But this makes it possible to make it a bit dynamic, when you can set the number in the settings.
imho default should be 7, so that a week resembles 1 cycle.(most of us do have a pattern in that cycle)

I'll take a look at your code tomorrow.
Thanks a lot for your time and effort.

oliexdev, 2020-08-18:
it's an exponentially smoothed moving average with 10% smoothing

[image]

why do you think its for temperature?

Marwel, 2020-08-18:
Because it came from https://www.fourmilab.ch/hackdiet/e4/feedback.html#FeedBackLab

oliexdev, 2020-08-18:
No the formula for that is
https://www.fourmilab.ch/hackdiet/e4/pencilpaper.html#PencilFeedbackLab

Marwel, 2020-08-18:
First link in that section is the url to what I sent. In that section he also describes the vars as temp, but nevertheless you are right, the formula can be used for a 1 day trend line.
The moving average is more accurate and in combination with the best fit trend at the bottom of that page they are great tools to predict where you're going with your weight.

oliexdev, 2020-08-18:
We can check with some test data if the Libra app use the expotenial
average calculation or moving average calculation with 7 days.

The prediction needs to be implemented but that should be not so difficult.

Marwel, 2020-08-18:
There was a setting in libra... How many days it should use for the trendline calculations. And also one for the best fit trend. Both were default set to 7.
If you increased the var in the setting, the line got smoother.

oliexdev, 2020-08-18:
If you have 4 measurements in Libra and you set the average frame to 7 then
it stills calculate the first values as trend values? Need to understand it
better.

Marwel, 2020-08-18:
For libra a day is also a measurement.
I assume that it takes all available measurements in that given time frame and divides it through the number of measurements found.

oliexdev, 2020-08-18:
Could be but if you increase the days and let the measurements constant
than the line gets smoother. Maybe they use all measurements that they are
found in the frame and divide it constant with the days....

oliexdev, 2020-08-19:
I implemented for testing the average moving formula with 7 days but I don't get the values like the libra app!?
It's somehow strange what they calculating. Could verify with the formula above the libra app calculation for the first values?

Marwel, 2020-08-19:
I'll try to convert my data for an import to libra.
With this, we should be able to make a better comparison.

Marwel, 2020-08-19:
OK, done.
Yeah the graph in the two apps is different.
[libra](https://lh3.googleusercontent.com/pw/ACtC-3dCO9JPEurs8hEZk7cxko7jcFf5jF_UkRinu7Fx88rPGjpkYu1n2HMjXjhxd5OXo0CIY-64rv-XqAEiMO0tlJQzzi4o0OJZV6BW3GxcwbgjkHOsgpJYQdTitwXXS_wLRocJ5Pwno9bm74u_GaelAXxcVg=w430-h906-no?authuser=0)
[openscale](https://lh3.googleusercontent.com/pw/ACtC-3dTffZe8PEAEYS7KPFJw62O69juDxewmO2vcBOF4odAg8zK3dIFL0xmJ8WoMNGAQer6E0d-AWWzGQQbxJ9TJ57iMiFH9p7Nj48OCiRTWf3iCkbNaMP7yZ6jwisF-x4_agErAruYxvNmGywybyGXzADvBA=w430-h906-no?authuser=0)

Marwel, 2020-08-20:
Just checked the code you altered and it seems right to me.
The curve looks better than the other one smoothed with 10%.
But it seems as if some trendline entries were not set. Or at least with an interesting value.

In the pic above of openscale, you should see on the 10.8. a more drastic increase and it looks like a direct line to the next entry.

oliexdev, 2020-08-20:
Yeah something is strange. Did you verified the calculation in the libra app with the pen & paper with the moving average formula? I get wrong values with this formula ...

Marwel, 2020-08-21:
Not yet. Right now i'm stuck in a project.
I'll try to sample a week or 2 of my data and hopefully on Sunday, i'll have time for that.

Marwel, 2020-08-26:
> Hi, I'd like to chime in, if you don't mind:
> Comparing Libra and openscale calculations: I don't think it is helpful to emulate Libras behaviour here. There is something weird (i.e. wrong) going on, if their "smooth" Trendline has a sharp change like on 6th of Aug. No one changes this fast from weight loss to weight gain.

This is actually my data of a rather interesting week/weekend.

> **TL;DR:** I would advocate to use the exponential moving average in openScale. It is easy to understand and has a sound mathematical underpinning. Emulating the behaviour of another app blindly does not serve any purpose imho.

Yeah, you're right. Emulating the same behaviour blindly is not a great idea.
The trendline is a great tool, also the prediction line is interesting.
Libra introduced them to me, so i figured, it would be good to have something resembling those features.

The graphs was just to show, that there are some minor issues, which i had no time up to now to investigate.
Both formulas are fairly simple regarding the math and @oliexdev made both work.

oliexdev, 2020-08-31:
> Comparing Libra and openscale calculations: I don't think it is helpful to emulate Libras behaviour here. There is something weird (i.e. wrong) going on, if their "smooth" Trendline has a sharp change like on 6th of Aug. No one changes this fast from weight loss to weight gain.

You are right the libra trendline use some other calculation which I don't really understand.

> The selling point: It is dead simple to calculate, you can even do it by hand. You just need the last trend value and your new weight (as @oliexdev mentioned).

The exponentially smoothed moving average calculation is more efficient and more cleaner to program. So I will stick with this formula until somebody has a better version.

> Yeah, you're right. Emulating the same behaviour blindly is not a great idea.
> The trendline is a great tool, also the prediction line is interesting.
> Libra introduced them to me, so i figured, it would be good to have something resembling those features.

I will implement the trendline and the prediction as an option if I have a bit time.

Marwel, 2020-09-09:
Looks good to me.

How did you get the prediction to be a curve?
I thought you would just get the derivative, but it seems you invested a little bit more on this.

Thank you for your time and effort on this.

oliexdev, 2020-09-13:
the prediction curve is a third order polynomial curve fit. I merged the result into the master branch.

closed 2020-09-13 by oliexdev