oliexdev/openScale #614: Unable to generate Bluetooth HCI Snoop Log on Android 8-10

DeeKey, 2020-08-17:
**Is your feature request related to a problem? Please describe.**
I am not able to generate  Bluetooth HCI Snoop Log needed to complete new scale support.
Logfile does not appear at all even though the option is turned on. Note that the phone is not rooted.

**Describe the solution you'd like**
I would like to get a clean instruction how to generate logs or a application to do it.
If phone must be rooted to get the log then it must be clearly stated in instructions.

**Describe alternatives you've considered**
Tried my old phone - it can generate Bluetooth HCI Snoop Log, but has only BT2.0 which is not working with my scales (v.4)
Also tried on

**Additional context**
Also tried to generate  Bluetooth HCI Snoop Log in uprooted Android 8 phone. Same problem -  log file is not available.

oliexdev, 2020-08-18:
That's a smartphone problem not a openScale problem. Maybe the HCI Snoop log file is saved in a different place?

closed 2020-08-18 by oliexdev

DeeKey, 2020-08-24:
It is not that trivial to get HCI Snoop log. I am still not able to do this.
At least worth mentioning that the process of taking HCI Snoop log differs from phone to phone in your readme file.
Also it is not clear for me if the phone need to be rooted.
https://stackoverflow.com/questions/28445552/bluetooth-hci-snoop-log-not-generated