oliexdev/openScale #618: Wrong Time/Date and info_step_on_scale is displayed

vonneudeck, 2020-08-30:
**Describe the bug**
When I step on the scale (Medisana BS444) sometimes I get the info_step_on_scale message. If that happens, it does take my values, but makes up time and date by using the date and time of the last measurement and add few minutes to the time of the last measurement. 

**To Reproduce**
1. Step on the scale
1. Do the measurement
1. Wait for the message to appear on the app
1. See the measurement with the wrong date.

Often the scale can’t be found and I have to disable and enable the connection within the app. Especially then the errors appear.

Reproduced with 2.3.1

**Expected behavior**
Use the actual date and time for the measurement.

**Additional context**
Seems to be connected with the display of the info message.

**Debug log**
I attached two logs. One with the faulty behaviour and one without.
[openScale_2020-08-30_18-40.txt](https://github.com/oliexdev/openScale/files/5147208/openScale_2020-08-30_18-40.txt)
[openScale_2020-08-30_18-32.txt](https://github.com/oliexdev/openScale/files/5147209/openScale_2020-08-30_18-32.txt)

oliexdev, 2020-08-31:
I am sorry I don't own this scale, without that scale its nearly impossible to solve that error.

closed 2020-09-13 by oliexdev

vonneudeck, 2020-09-13:
Sure, I get that.

What would be a good approach to debug with the scale?
I am willing to spend time on this, correcting the values manually every time is a pain.

I could also send it as a parcel to you, if you feel like doing it yourself.

oliexdev, 2020-09-13:
a good starting point is to read this https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale