oliexdev/openScale #620: Starting Openscale not possible

heavyweight1900, 2020-08-31:
When I try to start Openscale, the message "an unexpected failure happened " is shown. 

Here are the failure details:

Build version: 2.3.1-pro 
Current date: 2020-08-31 17:53:05 
Device: Samsung SM-A920F 

Stack trace:  
java.lang.RuntimeException: Exception while computing database live data.
at androidx.room.RoomTrackingLiveData$1.run(RoomTrackingLiveData.java:92)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
at java.lang.Thread.run(Thread.java:919)
Caused by: java.lang.IllegalStateException: Migration didn't properly handle: scaleMeasurements(com.health.openscale.core.datatypes.ScaleMeasurement).
Expected:
TableInfo{name='scaleMeasurements', columns={chest=Column{name='chest', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, thigh=Column{name='thigh', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, caliper2=Column{name='caliper2', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, weight=Column{name='weight', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, caliper3=Column{name='caliper3', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, bone=Column{name='bone', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, calories=Column{name='calories', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, neck=Column{name='neck', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, userId=Column{name='userId', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, water=Column{name='water', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, enabled=Column{name='enabled', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, hip=Column{name='hip', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, datetime=Column{name='datetime', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, lbm=Column{name='lbm', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, muscle=Column{name='muscle', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, caliper1=Column{name='caliper1', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, fat=Column{name='fat', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, waist=Column{name='waist', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, comment=Column{name='comment', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, id=Column{name='id', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=1, defaultValue='null'}, visceralFat=Column{name='visceralFat', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, biceps=Column{name='biceps', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}}, foreignKeys=[ForeignKey{referenceTable='scaleUsers', onDelete='CASCADE', onUpdate='NO ACTION', columnNames=[userId], referenceColumnNames=[id]}], indices=[Index{name='index_scaleMeasurements_userId_datetime', unique=true, columns=[userId, datetime]}]}
Found:
TableInfo{name='scaleMeasurements', columns={chest=Column{name='chest', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, thigh=Column{name='thigh', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, caliper2=Column{name='caliper2', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, weight=Column{name='weight', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, caliper3=Column{name='caliper3', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, bone=Column{name='bone', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, calories=Column{name='calories', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, neck=Column{name='neck', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, userId=Column{name='userId', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, water=Column{name='water', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, enabled=Column{name='enabled', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, hip=Column{name='hip', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, datetime=Column{name='datetime', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='null'}, lbm=Column{name='lbm', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, muscle=Column{name='muscle', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, caliper1=Column{name='caliper1', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, fat=Column{name='fat', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, waist=Column{name='waist', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, comment=Column{name='comment', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, id=Column{name='id', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=1, defaultValue='null'}, visceralFat=Column{name='visceralFat', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}, biceps=Column{name='biceps', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='null'}}, foreignKeys=[], indices=[]}
at androidx.room.RoomOpenHelper.onUpgrade(RoomOpenHelper.java:103)
at androidx.sqlite.db.framework.FrameworkSQLiteOpenHelper$OpenHelper.onUpgrade(FrameworkSQLiteOpenHelper.java:124)
at android.database.sqlite.SQLiteOpenHelper.getDatabaseLocked(SQLiteOpenHelper.java:493)
at android.database.sqlite.SQLiteOpenHelper.getWritableDatabase(SQLiteOpenHelper.java:391)
at androidx.sqlite.db.framework.FrameworkSQLiteOpenHelper$OpenHelper.getWritableSupportDatabase(FrameworkSQLiteOpenHelper.java:92)
at androidx.sqlite.db.framework.FrameworkSQLiteOpenHelper.getWritableDatabase(FrameworkSQLiteOpenHelper.java:53)
at androidx.room.RoomDatabase.inTransaction(RoomDatabase.java:476)
at androidx.room.RoomDatabase.assertNotSuspendingTransaction(RoomDatabase.java:281)
at androidx.room.RoomDatabase.query(RoomDatabase.java:324)
at androidx.room.util.DBUtil.query(DBUtil.java:83)
at com.health.openscale.core.database.ScaleMeasurementDAO_Impl$5.call(ScaleMeasurementDAO_Impl.java:846)
at com.health.openscale.core.database.ScaleMeasurementDAO_Impl$5.call(ScaleMeasurementDAO_Impl.java:843)
at androidx.room.RoomTrackingLiveData$1.run(RoomTrackingLiveData.java:90)
... 3 more

oliexdev, 2020-08-31:
there was something wrong with the database migration. From which version did you upgrade? To fix this problem you could uninstall an install openScale again. I hope you have a backup of your data.

closed 2020-08-31 by oliexdev

heavyweight1900, 2020-09-03:
Hi,

Thank you for your fast response.

I didn’t realize that I have made a database migration. 

I followed your suggestion and made several times a new installation of the Openscale app, but it always showed the same error. But at last the app works fine. 

The problem is now that I have the actual database with the new records from the scale and the database with the records of my backup.

Do you know a simple way to merge the two databases?

Kind regards

Werner