oliexdev/openScale #621: Unable to connect to Fitindex ES-26M

pinusc, 2020-09-03:
Hello,
I purchased a fitindex ED-26M because I saw it was on the supported scale list. My model looks exatly like the one pictured in the wiki. 
The scale won't show up in the bluetooth devices; I am also unable to pair with the Android bluetooth interface. 
I tried:
- connecting with openScale
- uninstalling and reinstalling openScale (making sure data was wiped)
- installing dev version of openScale

Then I tried pairing with the official app, which went smoothly. The scale still does not appear as a bluetooth device on android.

After delving within the logs of the official fitindex app, I believe I found the MAC address of the scale (which I was previously unable to even through linux, as I live in a very bluetooth-polluted environment). Attempting to pair/connect to the device on linux or android (which I could try after finding the correct MAC address) has been useless.  

I think the problem here is that openScale checks for the advertised name of the bluetooth scale; the expected name for this scale (as in BluetoothFactory.java) is "QN-scale"; however, my scale does not advertise any name, presenting itself only as a MAC address. 
In wireshark I can see some HCI events generated by this device, so I _think_ you don't need to pair/connect to the scale in order to receive events, and that's what even the official app does.

I know very little about how Bluetooth/BLE works; would it be possible for openScale to read events without pairing to the scale? Why did this use to work with the same model and now doesn't?

oliexdev, 2020-09-04:
please attach the openScale log file and please don't do a Bluetooth pairing with the Android system

closed 2020-09-19 by oliexdev

treisska, 2020-11-26:
Hello,

I'm having the exact same issue. The scale is recognized immediately by official application, but neither openscale, android standard Bluetooth stuff nor ble scanner can see it. 

My scale is branded as a Renpho ES-26M,but it look exactly the same. I've included logs from openscale and a screenshot of ble scanner. My scale is turned on and I'm  standing on it while doing my tests. 

I have the same result on my Mi 9t and a Samsung galaxy s7. 

Do you what is happening and how can it be fixed? 

[image]

`2020-11-26 19:29:50.060 Debug [2] AboutPreferences: Debug log enabled, openScale v2.3.2 (51), SDK 29, Xiaomi Mi 9T
2020-11-26 19:29:50.061 Debug [2] AboutPreferences: Selected user id(1) name(moi) birthday(Sun Nov 26 18:55:17 GMT+01:00 2000) age(20) body height(187.00) scale unit(kg) gender(male) initial weight(76.90) goal weight(74.00) goal date(Wed May 26 18:55:17 GMT+02:00 2021) measure unt(cm) activity level(0) assisted weighing(false)
2020-11-26 19:29:55.182 Info [2] BluetoothCentral: scan started
2020-11-26 19:29:55.222 Debug [2] BluetoothSettingsFragment: Found unsupported device [TV] Samsung 7 Series (65) [9C:8C:6E:56:AE:21]
2020-11-26 19:29:55.536 Debug [2] BluetoothSettingsFragment: Found unsupported device PHL-50PUS7303/12 [C4:98:5C:26:1F:BD]
2020-11-26 19:30:00.626 Debug [2] BluetoothSettingsFragment: Found unsupported device N04S7 [65:27:8F:A4:68:CB]
2020-11-26 19:30:05.756 Debug [2] BluetoothSettingsFragment: Found unsupported device Prodigio_DA172BCEDB9D [DA:17:2B:CE:DB:9D]
2020-11-26 19:30:15.189 Info [2] BluetoothCentral: scan stopped
2020-11-26 19:30:17.694 Info [2] BluetoothCentral: no scan to stop because no scan is running
`

EtzBetz, 2021-08-05:
I'm having the same problem as @canu1337. I have a Renpho scale, model name is es-26m-w and on the back it has also written "cs20m" as it's fcc-id.

This is the log:

```2021-08-05 13:04:06.449 Debug [2] AboutPreferences: Debug log enabled, openScale v2.3.5-light (54), SDK 30, Google Pixel 2
2021-08-05 13:04:06.449 Debug [2] AboutPreferences: Selected user id(1) name(Raphael) birthday(Sun Jun 01 00:00:00 GMT+02:00 1997) age(24) body height(200,00) scale unit(kg) gender(male) initial weight(97,00) goal weight(90,00) goal date(Fri Nov 05 00:00:00 GMT+01:00 2021) measure unt(cm) activity level(0) assisted weighing(false)
2021-08-05 13:05:35.390 Info [2] BluetoothCentral: scan started
2021-08-05 13:05:37.255 Debug [2] BluetoothSettingsFragment: Found unsupported device Mi Smart Band 5 [FC:4D:D0:CC:B2:20]
2021-08-05 13:05:55.399 Info [2] BluetoothCentral: scan stopped
2021-08-05 13:05:57.288 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-08-05 13:06:16.545 Info [2] BluetoothCentral: scan started
2021-08-05 13:06:16.937 Debug [2] BluetoothCentral: scanning timeout, restarting scan
2021-08-05 13:06:16.940 Info [2] BluetoothCentral: scan stopped
2021-08-05 13:06:17.948 Info [2] BluetoothCentral: scan started
2021-08-05 13:06:19.677 Debug [2] BluetoothSettingsFragment: Found unsupported device Mi Smart Band 5 [FC:4D:D0:CC:B2:20]
2021-08-05 13:06:36.553 Info [2] BluetoothCentral: scan stopped
2021-08-05 13:06:38.521 Info [2] BluetoothCentral: no scan to stop because no scan is running
```

I'm happy to help out somehow, I already tried digging through it with wireshark and started some guesswork, but im not sure how to continue.

AlphaDE, 2021-11-02:
Same issue here, OpenScale smoothly works with ES-CS20M (V1) but not with ES-26M.

natschil, 2022-08-30:
For the es-26-m model, here is a sample bluetooth packet. The f9 24 in the payload is the weight. 
[image]

Relevant to detecting this is probably also

BD_ADDR: ed:67:39:c6:1d:b5 (ed:67:39:c6:1d:b5)