oliexdev/openScale #622: Add build instructions to the readme or FAQ

jimhenshaw, 2020-09-04:
I'm unable to build your repository without assistance. I've tried forking it and using Android Studio to create a project from the repository, but it runs into problems. I changed the module to Android API on androidTest, main, and test. 

Next, to solve java.lang.ClassNotFoundException: com.google.wireless.android.sdk.stats.IntellijIndexingStats$Index in Android Studio I had to change the Java Compiler from javac to eclipse. (File-> Settings-> Build, Execution, Deployment-> Compiler -> Java Compiler)

This left me with the following error which I'm unable to resolve:
Error:Module 'androidTest' production and 1 more: java.lang.NoClassDefFoundError: com/android/resources/ResourceFolderType

If I had step by step instructions that I could follow, then I could hopefully avoid these issues. I tried reading the readme and FAQ for information, but didn't find anything relating to compiling the code on your own.

oliexdev, 2020-09-04:
Sorry but those building problems are not related to openScale. openScale don't even use com.google.wireless.android.sdk.stats

closed 2020-09-04 by oliexdev

jimhenshaw, 2020-09-04:
I was trying to relate the difficulty of building your software, and requesting that build instructions be created. The google stats package may not be used by your application and may have been an error in Android Studio's import project function. That's not the point though. The point is I can't build your software. Can you add build instructions?

oliexdev, 2020-09-05:
The build instruction is: GoTo AndroidStudio import the project and click on build! That's its...