oliexdev/openScale #626: Total body water, muscle and fat % not showing for Mi body composition 2

odyssoulis, 2020-09-14:
First up, thanks for the effort on developing the app!

**Describe the bug**
After successfully connecting the Mi body composition 2 to the app, i only see results about weight, BMI, BMR and TDEE.

However, total body water, Muscle% and Body fat% are **not** logged.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build)

**Additional context**
Not sure, if the issue is the dev build i'm running. Maybe downloading the app from PlayStore is a different build and this issue is resolved there?
Are these measurements, for the specific scale, even supported in the first place?

oliexdev, 2020-09-15:
please attach the openScale log file which is written in the issue template

odyssoulis, 2020-09-16:
Hey, actually it turns out only some measurements have this issue.

I will check if its something on my side, if i suspect otherwise i will re open the issue together with the log files!

closed 2020-09-16 by odyssoulis