oliexdev/openScale #627: Mi Body Composition Scale loses track of time?

bobobo1618, 2020-09-17:
**Describe the bug**
After several months, OpenScale ceases to pick up readings from the Xiaomi Mi Body Composition Scale. Upon syncing the scale with the Mi Fit app, new readings appear to be read.

Logs show measurements with invalid dates (1970), which are dropped by OpenScale:

```
2020-09-17 10:15:18.744 Error [2] BluetoothMiScale2: Invalid Mi scale weight year 1970
2020-09-17 10:15:18.877 Debug [2] BluetoothMiScale2: DataChange hex data: 02 24 B2 07 01 04 10 0B 20 FD FF 7C 38
```

So I suspect the issue might be that the scale needs periodic time updates from the app and defaults to 1970 if it doesn't receive them.

Alternatively, I've noticed that there are _a lot_ of previously-synced readings. Could it be internal memory is filling up?

**To Reproduce**
Steps to reproduce the behavior:
1. Set up Xiaomi Scale
2. Use with OpenScale for ~9 months

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): I haven't been using the latest build for the months I've owned this scale but I'm using it now.

**Expected behavior**
Continues to function

**Debug log**

[openScale_2020-09-17_10-13.txt](https://github.com/oliexdev/openScale/files/5237560/openScale_2020-09-17_10-13.txt)

oliexdev, 2020-09-19:
Using two apps is not recommended because Mi scale needs to set up an unique identifier. Please use either only the Mi Fit **or** the openScale app otherwise one app get confused :wink: 

Reading your log file everything looks good. You need to know that the mi scale never delete a measurement on the scale. What happened if the internal store is full? I don't know! Maybe it will stop working or it will overwrite past measurement (ring storage). Until now, I didn't find any delete function.

The invalid years measurements are such measurements which the initial time set was missing (maybe your first measurements without using first time the app).

The measurements with zero impedance are measurements were the impedance measurement didn't work (for e.g. measuring with socks on)

closed 2020-09-19 by oliexdev

bobobo1618, 2020-09-19:
I don't use two apps, I use OpenScale exclusively and only used Mi Fit once, for troubleshooting, after trying the OpenScale app several times and getting none of the new readings I knew were there.

There are about 30 days of data that I know are on the scale but OpenScale hasn't read.

I assume those days are the days with the 1970 reading.

oliexdev, 2020-09-19:
from the log file your last measurement was added:

```
2020-09-17 10:24:37.865 Debug [2] OpenScale: assign measurement to the nearest measurement with the user User1 (smartUserAssignment=on)
2020-09-17 10:24:37.867 Debug [2] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Thu Sep 17 10:24:00 GMT+02:00 2020, WEIGHT: 84.60, FAT: 40.83, WATER: 42.24, MUSCLE: 55.69, LBM: 0.00, WAIST: 0.00, HIP: 0.00, BONE: 2.94, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: 
```

But many measurements are thrown away because they were already added.
All in all you can see in the log file what has been happened with the measurements.

bobobo1618, 2020-09-19:
> from the log file your last measurement was added:

Yes, as I said in the initial report, after using the Mi Fit app once, OpenScale started to collect readings again. This is one of those readings.

> But many measurements are thrown away because they were already added.
> All in all you can see in the log file what has been happened with the measurements.

Here's what I know:

- There are ~40 readings missing from OpenScale:
  - The scale is used once a day, every day, unless on holiday
  - When I synced on the 17th September, I only received data until August 9th, there is a missing period between 2020.08.09 and 2020.09.17
  - When I used the scale a few times on the 17th of September and tried to sync, the readings did not show
  - When I used the scale after using the Mi Fit app, the readings started to show again and now consistently show
- The logs show:
  - 400 records that were thrown away due to being duplicates of existing records
  - 17 records that were thrown away due to the year being 1970
  - 6 records that were thrown away due to no user being selected
  - 6 records that were thrown away due to out of range measurements
  - 2 records that were added from the 17th of September

The main thing is that despite what the logs are saying, there are missing records. Something isn't working as expected. There is a bug somewhere.