oliexdev/openScale #633: Sanitas SBF 70 says 'Full' and user measurements are not being downloaded/recorded.

timoc, 2020-09-23:
**Describe the bug**
Sanitas SBF 70 says 'Full' and measurements are not being downloaded/recorded. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Openscale and connect with bluetooth.
2. Step on scale, and see measurements appear in openscale.
3. Wait for sale to settle, and start to display non body weight measurements
4. See Error on scale "Full" for user
5. See no measurements recorded in the Openscale for that user.

Reproduced with latest version from Fdroid.

**Expected behavior**
All measurements are downloaded from scale when sync. 
Option to 'clear entries on download'

**Additional context**
None.

**Debug log**
DO not have one to hand. Will update issue.

pedrom34, 2020-09-27:
Same issue for me.

marius1, 2020-10-04:
Have the same problem, I think deleting an entry when it has been downloaded should be the correct approach.

timoc, 2020-10-06:
I reset the scale as per the Wiki advice, but now i have a new/similar capacity problem - which is 'maximum number of users assigned" or similar.

melkor0, 2020-10-08:
@timoc : This seems to be a different issue. Try to connect to the scale when 'location data' (what you would do if trying to find a GPS signal) is activated. There is a functional overlap when you want to use BLE and when you want to use GPS / cell tower location / etc, you need to have the location function activated.

oliexdev, 2020-10-12:
> Sanitas SBF 70 says 'Full

So on the scale display it says "full"?

Could you attach the openScale debug log file?

Normally openScale should delete saved measurements on the scale. The log file should say `Deleting saved measurements for ...`

marius1, 2020-10-12:
In my case I stopped using the Health Coach a couple of months ago because it is really flaky. After a while the scale was full (doesn't show measurements except weight and the 'FULL' message). Recently I bought openScale to get the data again and initially it imported all the data from the scale, but at that moment did not delete it I guess. I reset the scale as well, now it is working again. Will get the debug log next time.

oliexdev, 2020-10-12:
indeed if you switch from one app to another it will not delete any measurements on the scale, this has to be done manually by resetting the scale.

closed 2020-10-12 by oliexdev

timoc, 2020-10-13:
> 
> 
> > Sanitas SBF 70 says 'Full
> 
> So on the scale display it says "full"?
> 
> Could you attach the openScale debug log file?
> 
> Normally openScale should delete saved measurements on the scale. The log file should say `Deleting saved measurements for ...`

I will try and create a log file. There were two erriors 
Error 1 is that the measurements are full.
Error 2 is that the number of available users was full.

I am still getting this user error after a while I think it may be some auto add feature?

timoc, 2020-11-10:
Process - take weight measurement, open openscale, turn on BT, connect to scale.
error "Max. number of concurrent scale users reached"
No weight recorded/downloaded.

Logfile:

```
2020-11-10 12:20:45.727 Debug [1] AboutPreferences: Debug log enabled, openScale v2.3.1 (50), SDK 28, HUAWEI STK-LX1
2020-11-10 12:20:45.731 Debug [1] AboutPreferences: Selected user id(1) name(Tim)
2020-11-10 12:20:52.057 Debug [1] OpenScale: Trying to connect to bluetooth device [80:30:DC:EA:4E:65] (SANITAS SBF70)
2020-11-10 12:20:52.065 Debug [1] BluetoothCommunication: No location permission, connecting without LE scan
2020-11-10 12:20:53.069 Debug [1] BluetoothCommunication: Try to connect to BLE device 80:30:DC:EA:4E:65
2020-11-10 12:20:53.073 Warning [1] BluetoothCentral: peripheral with address '80:30:DC:EA:4E:65' is not in the Bluetooth cache, hence connection may fail
2020-11-10 12:20:53.179 Info [1] BluetoothPeripheral: connect to 'null' (80:30:DC:EA:4E:65) using TRANSPORT_LE
2020-11-10 12:20:53.654 Info [19895] BluetoothPeripheral: connected to 'null' (BOND_NONE) in 0.4s
2020-11-10 12:20:53.657 Debug [1] BluetoothPeripheral: discovering services of 'null' with delay of 0 ms
2020-11-10 12:20:54.809 Info [19895] BluetoothPeripheral: discovered 5 services for 'null'
2020-11-10 12:20:54.811 Debug [1] BluetoothCommunication: connected to 'null'
2020-11-10 12:20:54.813 Debug [1] BluetoothCommunication: Step Nr 0
2020-11-10 12:20:54.814 Debug [1] BluetoothCommunication: Invoke set notification on 0xffe1
2020-11-10 12:20:54.815 Debug [1] BluetoothCommunication: Stop machine state
2020-11-10 12:20:54.817 Debug [1] BluetoothCommunication: Successful Bluetooth services discovered
2020-11-10 12:20:54.818 Debug [1] BluetoothCommunication: Resume machine state
2020-11-10 12:20:54.818 Debug [1] BluetoothCommunication: Step Nr 1
2020-11-10 12:20:54.820 Debug [1] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2020-11-10 12:20:54.821 Debug [1] BluetoothCommunication: Stop machine state
2020-11-10 12:20:54.832 Debug [1] MainActivity: Bluetooth connection successful established
2020-11-10 12:20:54.959 Debug [1] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2020-11-10 12:20:54.959 Debug [1] BluetoothCommunication: Resume machine state
2020-11-10 12:20:54.960 Debug [1] BluetoothCommunication: Step Nr 2
2020-11-10 12:20:54.962 Debug [1] BluetoothCommunication: Invoke write bytes [E9 5F AA 77 96] on 0xffe1
2020-11-10 12:20:54.963 Debug [1] BluetoothCommunication: Step Nr 3
2020-11-10 12:20:54.965 Debug [1] BluetoothCommunication: Invoke write bytes [E7 4F 00 00 00 00 00 00 00 00] on 0xffe1
2020-11-10 12:20:54.966 Debug [1] BluetoothCommunication: Stop machine state
2020-11-10 12:20:54.969 Debug [1] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.057 Debug [1] BluetoothCommunication: SUCCESS: Writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.061 Debug [1] BluetoothPeripheral: writing <e95faa7796> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.062 Debug [1] BluetoothBeurerSanitas: Got init ack from scale; scale is ready
2020-11-10 12:20:55.064 Debug [1] BluetoothCommunication: Resume machine state
2020-11-10 12:20:55.065 Debug [1] BluetoothCommunication: Step Nr 4
2020-11-10 12:20:55.066 Debug [1] BluetoothCommunication: Invoke write bytes [E7 33] on 0xffe1
2020-11-10 12:20:55.068 Debug [1] BluetoothCommunication: Stop machine state
2020-11-10 12:20:55.156 Debug [1] BluetoothCommunication: SUCCESS: Writing <E9 5F AA 77 96> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.162 Debug [1] BluetoothPeripheral: writing <e74f0000000000000000> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.255 Debug [1] BluetoothCommunication: SUCCESS: Writing <E7 4F 00 00 00 00 00 00 00 00> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.259 Debug [1] BluetoothPeripheral: writing <e733> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.360 Debug [1] BluetoothBeurerSanitas: Battery level: 36; threshold: weight=2.00, body fat=2.00; unit: 1; requested user: exists=false, has reference weight=false, has measurement=false; scale version: 5
2020-11-10 12:20:55.362 Debug [1] BluetoothCommunication: Resume machine state
2020-11-10 12:20:55.363 Debug [1] BluetoothCommunication: Step Nr 5
2020-11-10 12:20:55.364 Debug [1] BluetoothCommunication: Step Nr 6
2020-11-10 12:20:55.365 Debug [1] BluetoothBeurerSanitas: Create user: Tim
2020-11-10 12:20:55.368 Debug [1] BluetoothCommunication: Invoke write bytes [E7 31 00 00 00 00 00 00 00 65 54 49 4D 47 03 01 B6 81] on 0xffe1
2020-11-10 12:20:55.369 Debug [1] BluetoothCommunication: Stop machine state
2020-11-10 12:20:55.370 Debug [1] BluetoothCommunication: SUCCESS: Writing <E7 33> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.374 Debug [1] BluetoothPeripheral: writing <e731000000000000006554494d470301b681> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.457 Debug [1] BluetoothCommunication: SUCCESS: Writing <E7 31 00 00 00 00 00 00 00 65 54 49 4D 47 03 01 B6 81> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2020-11-10 12:20:55.655 Debug [1] BluetoothBeurerSanitas: Cannot create additional scale user (error 0x03)
2020-11-10 12:20:55.656 Debug [1] BluetoothBeurerSanitas: Send disconnect command to scale
[... cut ...]
2020-11-10 12:22:00.717 Debug [1] OpenScale: Disconnecting from bluetooth device
2020-11-10 12:22:00.718 Debug [1] BluetoothCommunication: Bluetooth disconnect
2020-11-10 12:22:00.718 Info [1] BluetoothCentral: no scan to stop because no scan is running
2020-11-10 12:22:00.719 Error [1] BluetoothCentral: cannot cancel connection to unknown peripheral 80:30:DC:EA:4E:65
~                                                                                                                                    
```