oliexdev/openScale #635: Support Beurer BF 105

MiNGlKLW, 2020-10-03:
Beurer BF 105 
https://www.beurer.com/web/de/produkte/wellbeing/gewicht-und-diagnose/diagnosewaagen/bf-105.php
Beurer HealthManager from Google Play / App Store

_To support a new scale it is necessary to gather some information._

**Step 1: Read the general reverse engineer process**

- [x] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

user settings in the vendors app: sex : male;  body height: 180cm; age: 48; activity level: 3;

1. Bluetooth HCI Snoop log file
measured true values in the vendors app for the 1. HCI Snoop log file:
btsnoop_hci_01.log
17.06.2020 18h51
weight: 79,4kg
body fat 19,7%
H2O: 52,3%
muscle: 38,0%
bones:3,3kg
BMR: 1689kcla
AMR 2534Kcal

_--> Attach the 1. HCI Snoop log file here <--_
[btsnoop_hci_01.log](https://github.com/oliexdev/openScale/files/5321587/btsnoop_hci_01.log)

```
2. Bluetooth HCI Snoop log file
measured true values in the vendors app for the 2. HCI Snoop log file:
btsnoop_hci_02.log
17.06.2020 18h53
weight: 85,0 kg
body fat 22,1 %
H2O: 50,5 %
muscle: 36,7%
bones: 3,4kg
BMR: 1745 kcla
AMR 2618 kcal
_--> Attach the 2. HCI Snoop log file here <--_
[btsnoop_hci_02.log](https://github.com/oliexdev/openScale/files/5321589/btsnoop_hci_02.log)

3. Bluetooth HCI Snoop log file
measured true values in the vendors app for the 3. HCI Snoop log file:
btsnoop_hci_03.log
17.06.2020 18h55
weight: 82,1 kg
body fat 20,8 %
H2O: 51,5 %
muscle: 37,3 %
bones: 3,4kg
BMR: 1716 kcla
AMR 2574 kcal
_--> Attach the 3. HCI Snoop log file here <--_
[btsnoop_hci_03.log](https://github.com/oliexdev/openScale/files/5321590/btsnoop_hci_03.log)

**Step 3: Discover Bluetooth services and characteristic**
_Read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#2-find-out-the-bluetooth-services-and-characteristic) how to create the openScale debug file_

_--> Attach the openScale debug log file here <--_
[openScale_2020-06-15_16-12.txt](https://github.com/oliexdev/openScale/files/5321591/openScale_2020-06-15_16-12.txt)
[openScale_2020-06-15_16-50.txt](https://github.com/oliexdev/openScale/files/5321593/openScale_2020-06-15_16-50.txt)

jensMF, 2021-04-02:
I am interested in this too. I plan to buy this.
@MiNGlKLW you accidentally wrote the part after the first Bluetooth HCI Snoop log file as codeblock. Maybe you could change that so the Links to the files are better usable.

If there is more data needed I am willing to help as soon as my device arrived.

MiNGlKLW, 2021-04-04:
@jensMF: thanks for the hint and I'm happy that someone else is interested in getting this model supported. I correceted my comment above. The licks should be usable now.

jensMF, 2021-04-09:
I now have the BF105 here and did some analyses and research.
I can not find any of the values in the hci snoop files with wireshark.
What I did find with an app called nRF Connect:
```
Generic Access
UUID: 0x1800
PRIMARY SERVICE

    Device Name
    UUID: 0x2A00
    Properties: READ
    Parsed value: BF105
    Raw value: (0x) 42-46-31-30-35, "BF105"

    Appearance
    UUID: 0x2A01
    Properties: READ
    Parsed value: [0] Unknown
    Raw value: (0x) 00-00

    Peripheral Privacy Flag
    UUID: 0x2A02
    Properties: READ, WRITE
    Parsed value: Privacy is disabled in this device
    Raw value: (0x) 00

    Reconnection Address
    UUID: 0x2A03
    Properties: WRITE

   Peripheral Preferred Connection Parameters
    UUID: 0x2A04
    Properties: READ
    Parsed value: Connection Interval: 100.00ms - 200.00ms,
    Slave Latency: 0,
    Supervision Timeout Multiplier: 1000
    Raw value: (0x) 50-00-A0-00-00-00-E8-03

Generic Attribute
UUID: 0x1801
PRIMARY SERVICE

    Service Changed
    UUID: 0x2A05
    Properties: INDICATE
    Descriptors:
    Client Characteristics Configuration
    UUID: 0x2902
    Value: Notifications and indications enabled or Value: Indications enabled

Device Information
UUID: 0x180A
PRIMARY SERVICE

    System ID
    UUID: 0x2A23
    Properties: READ
    Parsed value: <I prefer to keep this private>
    Raw value: <I prefer to keep this private>

    Model Number String
    UUID: 0x2A24
    Properties: READ
    Parsed value: BF105
    Raw value: (0x) 42-46-31-30-35, "BF105"

    Serial Number String
    UUID: 0x2A25
    Properties: READ
    Parsed value: Serial Number
[... 331 lines cut ...]
    UUID: 0000ffc1-0000-1000-8000-00805f9b34fb
    Properties: NOTIFY, WRITE, WRITE NO RESPONSE
    Descriptors:
    Client Characteristics Configuration
    UUID: 0x2902
    Value: Notifications and indications disabled or Value: Indications enabled
    Characteristic User Description
    UUID: 0x2901
    Value: Target Weight

    Img Block
    UUID: 0000ffc2-0000-1000-8000-00805f9b34fb
    Properties: NOTIFY, WRITE, WRITE NO RESPONSE
    Descriptors:
    Client Characteristics Configuration
    UUID: 0x2902
    Value: Notifications and indications disabled or Value: Indications enabled
    Characteristic User Description
    UUID: 0x2901
    Value: Target Weight
```

I hope this helps somehow.

I would really like to get advice how to proceed and, may be, get some important information. Help very appreciated. :-)

jensMF, 2021-04-12:
I compared some data and found at least the body fat value.
Here are all packages from @MiNGlKLW  's log files:
In btsnoop_hci_01.log, it is the `c5` in the value of this package:
```
No.     Time           Source                Destination           Protocol Length Info
    216 37.757367      90:9a:77:23:22:af (BF105) localhost ()          ATT      26     Rcvd Handle Value Indication, Handle: 0x004f (Unknown: Unknown)

Frame 216: 26 bytes on wire (208 bits), 26 bytes captured (208 bits)
Bluetooth
    [Source: 90:9a:77:23:22:af]
    [Destination: 00:00:00:00:00:00]
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: ACL Data (0x02)
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
    Length: 17
    CID: Attribute Protocol (0x0004)
Bluetooth Attribute Protocol
    Opcode: Handle Value Indication (0x1d)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1101 = Method: Handle Value Indication (0x1d)
    Handle: 0x004f (Unknown: Unknown)
        [Service UUID: 0000181b00001000800000805f9b34fb]
        [UUID: 00002a9c00001000800000805f9b34fb]
    Value: 9803c500a01b7c013c2f74205212
```
here is the hex dump too:
```
0000   02 40 20 15 00 11 00 04 00 1d 4f 00 98 03 c5 00
0010   a0 1b 7c 01 3c 2f 74 20 52 12
```
In btsnoop_hci_02.log, it is the `dd` in the value of this package:
```
No.     Time           Source                Destination           Protocol Length Info
    221 43.475863      90:9a:77:23:22:af (BF105) localhost ()          ATT      26     Rcvd Handle Value Indication, Handle: 0x004f (Unknown: Unknown)

Frame 221: 26 bytes on wire (208 bits), 26 bytes captured (208 bits)
Bluetooth
    [Source: 90:9a:77:23:22:af]
    [Destination: 00:00:00:00:00:00]
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: ACL Data (0x02)
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
    Length: 17
    CID: Attribute Protocol (0x0004)
Bluetooth Attribute Protocol
    Opcode: Handle Value Indication (0x1d)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1101 = Method: Handle Value Indication (0x1d)
    Handle: 0x004f (Unknown: Unknown)
        [Service UUID: 0000181b00001000800000805f9b34fb]
        [UUID: 00002a9c00001000800000805f9b34fb]
    Value: 9803dd00861c6f0114318621e411
```
here is the hex dump:
```
0000   02 40 20 15 00 11 00 04 00 1d 4f 00 98 03 dd 00
0010   86 1c 6f 01 14 31 86 21 e4 11
```
In btsnoop_hci_03.log, it is the `d0` in the value of this package:
```
No.     Time           Source                Destination           Protocol Length Info
    215 17.222843      90:9a:77:23:22:af (BF105) localhost ()          ATT      26     Rcvd Handle Value Indication, Handle: 0x004f (Unknown: Unknown)

Frame 215: 26 bytes on wire (208 bits), 26 bytes captured (208 bits)
Bluetooth
    [Source: 90:9a:77:23:22:af]
    [Destination: 00:00:00:00:00:00]
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: ACL Data (0x02)
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
    Length: 17
    CID: Attribute Protocol (0x0004)
Bluetooth Attribute Protocol
    Opcode: Handle Value Indication (0x1d)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1101 = Method: Handle Value Indication (0x1d)
    Handle: 0x004f (Unknown: Unknown)
        [Service UUID: 0000181b00001000800000805f9b34fb]
        [UUID: 00002a9c00001000800000805f9b34fb]
    Value: 9803d000111c750124300821ee11
```
here is the hex dump:
```
0000   02 40 20 15 00 11 00 04 00 1d 4f 00 98 03 d0 00
0010   11 1c 75 01 24 30 08 21 ee 11
```

jensMF, 2021-04-12:
I have captured logs for two users with two devices.

**General values in the app for user 1:**

initialia: FE
sex : male
body height: 180cm
date of birth: I would prefer to keep that private.
activity level: 3
target weight: 76.0 kg

Here are my logs for this user:
**log 1:**
[Filtered_BTsnoop_log_1309_anon.log](https://github.com/oliexdev/openScale/files/6299997/Filtered_BTsnoop_log_1309_anon.log)
**values:**
05.01.2000 13:09 (here was an error, so the target weight might not be transmitted as well as the reference weight.)
weight: 89.3 kg
BMI: 27.5
body fat 24.6%
water: 54.2%
muscles: 41.0%
bones: 3.5 kg
BMR: 1868 Kcal
AMR: 2801 Kcal
upper boddy:
  fat: 24.2%
  muscles: 38.7%
under boddy:
  fat: 25.0%
  muscles: 43.2%

**log 2:**
[Filtered_BTsnoop_log_1535_anon.log](https://github.com/oliexdev/openScale/files/6300039/Filtered_BTsnoop_log_1535_anon.log)
**values:**
07.04.2021 15:35
reference weight: 81.2 kg
weight change: 0.0 kg
weight: 81.2 kg
BMI: 25.0
body fat 16.5%
water: 54.9%
muscles: 40.9%
bones: 3.5 kg
BMR: 1787 Kcal
AMR: 2681 Kcal
upper boddy:
  fat: 16.1%
  muscles: 38.6%
under boddy:
  fat: 17.0%
  muscles: 43.1%

**log 3:**
[Filtered_BTsnoop_log_1540_anon.log](https://github.com/oliexdev/openScale/files/6300079/Filtered_BTsnoop_log_1540_anon.log)
**values:**
07.04.2021 15:40
reference weight: 81.2 kg
weight change: 0.2 kg
weight: 87.4 kg
BMI: 26.9
body fat 19.2%
water: 52.6%
muscles: 39.2%
bones: 3.7 kg
BMR: 1849 Kcal
AMR: 2774 Kcal
upper boddy:
  fat: 18.5%
  muscles: 37.2%
under boddy:
  fat: 20.0%
  muscles: 41.0%

**log 4:**
[Filtered_BTsnoop_log_1544_anon.log](https://github.com/oliexdev/openScale/files/6300098/Filtered_BTsnoop_log_1544_anon.log)
**values:**
07.04.2021 15:44
reference weight: 81.2 kg
weight change: -5.0 kg
weight: 82.3 kg
BMI: 25.4
body fat 16.9%
water: 54.7%
muscles: 40.7%
bones: 3.6 kg
BMR: 1799 Kcal
AMR: 2699 Kcal
upper boddy:
  fat: 16.3%
  muscles: 38.7%
under boddy:
  fat: 17.5%
  muscles: 42.6%

**log 5:**
[Filtered_BTsnoop_log_1547_anon.log](https://github.com/oliexdev/openScale/files/6300106/Filtered_BTsnoop_log_1547_anon.log)
**values:**
07.04.2021 15:47
reference weight: 81.2 kg
weight change: 2.4 kg
weight: 84.8 kg
BMI: 26.1
body fat 18.3%
water: 53.1%
muscles: 39.6%
bones: 3.6 kg
BMR: 1823 Kcal
AMR: 2735 Kcal
upper boddy:
  fat: 17.8%
  muscles: 37.3%
under boddy:
  fat: 18.9%
  muscles: 41.7%

**General values in the app for user 2:**

initialia: TN
sex : male
body height: 190cm
date of birth: 10.04.1996
activity level: 3
target weight: 77.6 kg

Here are my logs for this user:

**log 6:**
[Filtered_BTsnoop_log_TN_2328_anon.log](https://github.com/oliexdev/openScale/files/6300122/Filtered_BTsnoop_log_TN_2328_anon.log)
**values:**
10.04.2021 23:28
reference weight: -
weight change: -
weight: 84.9 kg
BMI: 23.5
body fat 13.3%
water: 58.9%
muscles: 44.1%
bones: 3.8 kg
BMR: 1917 Kcal
AMR: 3547 Kcal
upper boddy:
  fat: 12.7%
  muscles: 42.6%
under boddy:
  fat: 14.0%
  muscles: 45.6%

**log 7**:
[Filtered_BTsnoop_log_TN_2331_anon.log](https://github.com/oliexdev/openScale/files/6300147/Filtered_BTsnoop_log_TN_2331_anon.log)
**values:**
10.04.2021 23:31
reference weight: 84.9 kg
weight change: -2.9 kg
weight: 82.0 kg
BMI: 22.7
body fat 12.3%
water: 60.3%
muscles: 45.1%
bones: 3.7 kg
BMR: 1888 Kcal
AMR: 3493 Kcal
upper boddy:
  fat: 11.9%
  muscles: 43.0%
under boddy:
  fat: 12.7%
  muscles: 47.1%

**log 8:**
[Filtered_BTsnoop_log_TN_2336_anon.log](https://github.com/oliexdev/openScale/files/6300165/Filtered_BTsnoop_log_TN_2336_anon.log)
**values:**
10.04.2021 23:36
reference weight: 84.9 kg
weight change: +2.3 kg
weight: 84.3 kg
BMI: 23.3
body fat 13.3%
water: 58.6%
muscles: 43.9%
bones: 3.8 kg
BMR: 1911 Kcal
AMR: 3536 Kcal
upper boddy:
  fat: 12.8%
  muscles: 42.3%
under boddy:
  fat: 13.9%
  muscles: 45.5%

jensMF, 2021-04-12:
May be you could help us @oliexdev ?

jensMF, 2021-04-13:
I found out some more details:

this time, I was looking in my 7th log ([Filtered_BTsnoop_log_TN_2331_anon.log](https://github.com/oliexdev/openScale/files/6300147/Filtered_BTsnoop_log_TN_2331_anon.log)):

The BMI is in this package, it is little endian, and in this case is 3e00. It is in the 4th and 3rd last byte (or in the 4th and 3rd byte when one starts counting form the right side):
```No.     Time           Source                Destination           Protocol Length Info
    454 134.444272     b8:80:4f:40:92:2c (BF105) localhost ()          ATT      27     Rcvd Handle Value Indication, Handle: 0x0047 (Unknown: Unknown)

Frame 454: 27 bytes on wire (216 bits), 27 bytes captured (216 bits)
Bluetooth
    [Source: b8:80:4f:40:92:2c]
    [Destination: 00:00:00:00:00:00]
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: ACL Data (0x02)
Bluetooth HCI ACL Packet
    .... 0000 0000 0100 = Connection Handle: 0x004
    ..10 .... .... .... = PB Flag: First Automatically Flushable Packet (2)
    00.. .... .... .... = BC Flag: Point-To-Point (0)
    Data Total Length: 22
    Data
    [Connect in frame: 246]
    [Source BD_ADDR: b8:80:4f:40:92:2c]
    [Source Device Name: BF105]
    [Source Role: Unknown (0)]
    [Destination BD_ADDR: 00:00:00:00:00:00]
    [Destination Device Name: ]
    [Destination Role: Unknown (0)]
    [Current Mode: Unknown (-1)]
Bluetooth L2CAP Protocol
    Length: 18
    CID: Attribute Protocol (0x0004)
Bluetooth Attribute Protocol
    Opcode: Handle Value Indication (0x1d)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1101 = Method: Handle Value Indication (0x1d)
    Handle: 0x0047 (Unknown: Unknown)
        [Service UUID: 0000181d00001000800000805f9b34fb]
        [UUID: 00002a9d00001000800000805f9b34fb]
    Value: 0e1040e507040a171f2304e3006c07
```
here is the hex dump:
```
0000   02 04 20 16 00 12 00 04 00 1d 47 00 0e 10 40 e5
0010   07 04 0a 17 1f 23 04 e3 00 6c 07
```

Here are the body fat (7b00, directly after the generic 9803 part) and the muscle percentage (c301) (both in little endian):
```
No.     Time           Source                Destination           Protocol Length Info
    456 134.444919     b8:80:4f:40:92:2c (BF105) localhost ()          ATT      26     Rcvd Handle Value Indication, Handle: 0x004f (Unknown: Unknown)

Frame 456: 26 bytes on wire (208 bits), 26 bytes captured (208 bits)
Bluetooth
    [Source: b8:80:4f:40:92:2c]
    [Destination: 00:00:00:00:00:00]
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: ACL Data (0x02)
Bluetooth HCI ACL Packet
    .... 0000 0000 0100 = Connection Handle: 0x004
    ..10 .... .... .... = PB Flag: First Automatically Flushable Packet (2)
    00.. .... .... .... = BC Flag: Point-To-Point (0)
    Data Total Length: 21
    Data
    [Connect in frame: 246]
    [Source BD_ADDR: b8:80:4f:40:92:2c]
    [Source Device Name: BF105]
    [Source Role: Unknown (0)]
[... 114 lines cut ...]
>     Characteristic User Description
>     UUID: 0x2901
>     Value: Initials
> 
>     Target Weight
>     UUID: 00000003-0000-1000-8000-00805f9b34fb
>     Properties: READ, WRITE
>     Parsed value: (0x) FF-00
>     Raw value: (0x) FF-00
>     Descriptors:
>     Characteristic User Description
>     UUID: 0x2901
>     Value: Target Weight
> 
>     Activity Level
>     UUID: 00000004-0000-1000-8000-00805f9b34fb
>     Properties: READ, WRITE
>     Parsed value: (0x) FF
>     Raw value: (0x) FF
>     Descriptors:
>     Characteristic User Description
>     UUID: 0x2901
>     Value: Activity Level

But I could not find any of the other measured values.

jensMF, 2021-04-13:
I have made huge progress today (or tonight)!
I added the BF105 to the BluetoothFactory and used the BluetoothStandardWeightProfile:
```
if (deviceName.equals("BF105")) {
      return new BluetoothStandardWeightProfile(context);
}
```
and changed the hard coded CURRENT_USER_CONSENT to the Code of my user.
It nearly works:
The weight seems to be correct (83.4 Kg on the scale and 83.42 Kg in openScale). I could not check the BMI but when I calculate it it would be 25.7 with 83.4 and 25.8 with 83.42 and openScale shows: 25.75!
Muscle is 39.7% on the scale and 39.70% in open scale! :-)
Body fat is 18.90% on both and correct, too!

Now the differences:
Total body water: 52.8% on the scale and 44.04% in openScale (but as I never found that value in the snooped logs, it might be, that they send another value than they show)
Bone mass: 3.5 kg on the scale and 0.0 in openScale
BMR: 1809 kCal on the scale and 1905 kCal in openScale
AMR/TDEE (is that the same?): 2532 kCal on the scale and 2619 kCal in openScale

The upper and under body features do not exist in openScale.

jensMF, 2021-04-13:
What I realized just now:
For every time I stood on the scale to test it, there where added two measurements to openScale (with slightly different values.)

MiNGlKLW, 2021-04-14:
Hi jensMF, welcome! It's great to have anoter combatant who is interested in this scale!

> I added the BF105 to the BluetoothFactory and used the BluetoothStandardWeightProfile:
> 
> ```
> if (deviceName.equals("BF105")) {
>       return new BluetoothStandardWeightProfile(context);
> }
> ```
> 
This is interesting. I also tried exactly this but did not come as far as you did. 

When I connet to the scane, I get the following on the Android Studio console:
_2021-04-14 17:34:53.011 6736-6736/com.health.openscale D/BluetoothCommunication: Step Nr 7
2021-04-14 17:34:53.012 6736-6736/com.health.openscale D/BluetoothCommunication: Invoke delayed disconnect in 60s
2021-04-14 17:34:53.019 6736-6736/com.health.openscale D/BluetoothPeripheral: writing <02034520> to characteristic <00002a9f-0000-1000-8000-00805f9b34fb>
2021-04-14 17:34:53.105 6736-6736/com.health.openscale E/BluetoothStandardWeightProfile: **Not authorized**
2021-04-14 17:34:53.113 6736-6736/com.health.openscale D/BluetoothCommunication: SUCCESS: Writing <02 03 45 20> to <00002a9f-0000-1000-8000-00805f9b34fb>
2021-04-14 17:34:53.113 6736-6736/com.health.openscale D/BluetoothCommunication: Step Nr 7
2021-04-14 17:34:53.114 6736-6736/com.health.openscale D/BluetoothCommunication: Invoke delayed disconnect in 60s
2021-04-14 17:34:53.694 6736-6736/com.health.openscale E/BluetoothStandardWeightProfile: **ERROR: could not register new user**_ 

How did you enter the USER CONSENT in your code? If its e.g. 1234 in the scale did you then set CURRENT_USER_CONSENT = 1234; ? That's what I tried.

How do you get the measurement? Is the scale sending them when a measurement is done?

jensMF, 2021-04-17:
Hey @MiNGlKLW yes I put the code from the scale (let's say 1234) in the CURRENT_USER_CONSENT variable, like you did.
And I created an user in openScale where I put in the initials from the scale as name and the birthday, activity level, gender, measurement unit, height, initial weight and goal wight to the same values as for the user in the scale.
I think the most important part was the initials as name.

oliexdev, 2021-04-18:
see https://github.com/oliexdev/openScale/issues/502#issuecomment-620442379

jensMF, 2021-04-19:
@oliexdev Thank you for your answer. I am thinking about it and would start to implement the consent code.
Since I am completely new to this project may I ask you some questions:
  1. Where should I look to add the consent code to the user model? May be you could tell me the class name(s)?
  2. Where should I implement the user input part? The class name(s) would really help me here to.
 Thanks in advance and I hope I can get the support for these kind of scales ready.

oliexdev, 2021-04-22:
@jensMF see https://github.com/oliexdev/openScale/issues/681#issuecomment-756703956

jensMF, 2021-05-28:
I added a WIP pull request so you and other interested devs can see, what I have done already and what is missing. And to discuss the implementation I chose.
Here is the pull request: #735

krisjans, 2021-07-22:
> As you can see, my main problem is the UI. I have never really done UI stuff.

@jensMF maybe you could check my UI solution in #753 ([bccf0040f5f20e2842f8d5014675c9536edcc1c0](https://github.com/oliexdev/openScale/pull/753/commits/bccf0040f5f20e2842f8d5014675c9536edcc1c0)) and re-use it for BF105?
It seems that some-kind of Beurer middle-layer class will be needed to re-use Beurer specific code for BF600, BF850, BF950 and BF105.

krisjans, 2021-08-24:
Anyone willing to help out and test latest changes for #753 on BF105?
Latest automatic debug build is availabe here: openScale-2.3.5-debug.apk.
I'm especially interested if user initials is now stored and displayed on BF105 when you create new user from openScale.

p.s. @MiNGlKLW and @jensMF if you still have access to BF105, maybe you could help me out and test this version?

jensMF, 2021-09-10:
I started testing today.

First of all, I tried to connect to an existing scale user. The app crashed immediately.
Stack trace:
```
Build version: 2.3.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-09-09 20:29:37 
Device: Motorola XT1685 
OS version: Android 8.1.0 (SDK 27) 
 
Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{9da0942} (b7bd22c3-fb44-4db5-af87-c4a58ace0318)} not attached to a context.
	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:205)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6499)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
```
and debug log: 
[openScale_2021-09-09_20-21.txt](https://github.com/oliexdev/openScale/files/7140355/openScale_2021-09-09_20-21.txt)

What I noticed in the debug log is that a new user gets registered anyways and the consent code for that new user is used instead of requesting the correct consent code from the user.

jensMF, 2021-09-10:
After a restart, the scale user list was shown again, but the scale user I chose in the first run (index 10) was not present. When I selected another scale user (index 8), the consent codes of both are getting requested.

Debug log:
[openScale_2021-09-09_20-49.txt](https://github.com/oliexdev/openScale/files/7140372/openScale_2021-09-09_20-49.txt)

As you can see, first of all, the consent code from the first run gets used, then the correct consent code for user 8 is saved but then overwritten with the consent code of user 10.

When I reconnect the scale now, the scale user list is shown to me every time.

krisjans, 2021-09-10:
@jensMF Thanks for testing my debug version on BF105!
The crash you encountered seems to be the same as in this bug: #761 
My scale-user-selection dialog code expected that maximum user count on scale is 8, but on BF105 it is 10. Need to fix that.

jensMF, 2021-09-10:
The main problem is another one. I write you in the pull request.

dynaxo, 2021-09-17:
> Anyone willing to help out and test latest changes for #753 on BF105?

Hi, I just tested your debug build with my BF105 with the following results:
- Added a new user to openScale and connected it to an already existing user on my scale. Worked great and even imported all past weighing results
- Added a new user to openScale and let it create a new user on the scale. This worked but I had the impression that it did not take the right name (open scale: test, BF105: t). 
- Added a new user to openScale and let it create a new user on the scale. After that nothing happened. I disconnected and connected bluetooth again. This time it asks me for a consent code for user 5. I already have 4 users on my scale so the new user should have been 5. But according to my scale user 5 has not been created. When I cancel this dialog OpenScale asks me if I want to connect to a new user or if I want to create a new user again. So whatever I do I cannot create a new user from openScale.

This is just for your information. No need to debug this issue for me since my first user is still working fine.

krisjans, 2021-09-17:
> Hi, I just tested your debug build with my BF105 with the following results:
>
> * Added a new user to openScale and connected it to an already existing user on my scale. Worked great and even imported [...]

@dynaxo Thanks for your feedback!
Would you like to enable debug log in openScale app (see [here](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-%28FAQ%29#how-to-get-the-openscale-debug-log-file) how to do this) and try to create new user on the scale again? It would help me a lot to fix possible errors in BF105 support. Thanks in advance!

dynaxo, 2021-09-17:
Of course:
[openScale_2021-09-17_18-42.txt](https://github.com/oliexdev/openScale/files/7187437/openScale_2021-09-17_18-42.txt)
To force the user dialog to show up I deactivated and reactivated bluetooth. 
Thanks for your effort. Since the Beurer app does not work on my mobile this is the only option for me to use the scale with it.

krisjans, 2021-09-19:
> Of course:
> [openScale_2021-09-17_18-42.txt](https://github.com/oliexdev/openScale/files/7187437/openScale_2021-09-17_18-42.txt)
> To force the user dialog to show up I deactivated and reactivated bluetooth.

@dynaxo Thanks for the log!
This time, according to the log, user P-05 (initials `B  `),  was created successfully on the scale. Could you check the scale and see if user with index 5 exists?

dynaxo, 2021-09-21:
According to my scale there is no user 5. It shows the same new user dialog like for the other unused users 6 to 10.

krisjans, 2021-09-22:
> According to my scale there is no user 5. It shows the same new user dialog like for the other unused users 6 to 10.

@dynaxo Thanks for the response! It seems that there is some bug when creating new users on BF105.
Would you like to enable Bluetooth sniffing/logging and create scale-user form OpenSclae again? And provide both - btsnoop_hci.log and corresponding openScale debug log?

krisjans, 2021-09-22:
@jensMF Hello! When you tested #753 on BF105, did you tried to create new user on scale?

oliexdev, 2021-10-10:
see #753

closed 2021-10-10 by oliexdev

jensMF, 2021-11-08:
> @jensMF Hello! When you tested #753 on BF105, did you tried to create new user on scale?

@krisjans I thought I did. But I am not sure anymore. I wanted to debug this, too. I think I had this working in my fork.

jensMF, 2021-12-07:
I have a working version here: https://github.com/jensMF/openScale/tree/fix_no_user_on_scale_when_registerNewUser but I am not sure if it affects other scales using standard weight profile.
I am not sure, how good my solution is, too.
The problem is, that the writeBytes here (in this function): https://github.com/oliexdev/openScale/blob/7b1d92665a85bc1a02d53bb0eda3e94030ee7c62/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothStandardWeightProfile.java#L190
is successful at different times. I thought it would have been successful when we get the answer (that we process here): https://github.com/oliexdev/openScale/blob/7b1d92665a85bc1a02d53bb0eda3e94030ee7c62/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothStandardWeightProfile.java#L276-L281
but this it not the case.
It must be successful before we set the userdata here: https://github.com/oliexdev/openScale/blob/7b1d92665a85bc1a02d53bb0eda3e94030ee7c62/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothStandardWeightProfile.java#L193-L197
otherwise the new user only has default values and a consent code that we don't know.
If it is not successful before we request the reference measurement here: https://github.com/oliexdev/openScale/blob/7b1d92665a85bc1a02d53bb0eda3e94030ee7c62/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothStandardWeightProfile.java#L198-L203
no new user gets created on the scale.

The same is true for the userdata. If it is not set before we request the reference measurement the new user only has default values (except of the consent code, I think).

So I added handling of CHARACTERISTIC_CHANGE_INCREMENT to resume the machine state here: https://github.com/jensMF/openScale/blob/bffd0927082fdcb22f97d8173316220499bd051f/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothStandardWeightProfile.java#L260-L263

and used it here:
https://github.com/jensMF/openScale/blob/bffd0927082fdcb22f97d8173316220499bd051f/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothStandardWeightProfile.java#L192-L195
and here:
https://github.com/jensMF/openScale/blob/bffd0927082fdcb22f97d8173316220499bd051f/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothStandardWeightProfile.java#L200-L203

Instead we could override this function https://github.com/oliexdev/openScale/blob/7b1d92665a85bc1a02d53bb0eda3e94030ee7c62/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothCommunication.java#L468-L477
and add the handling when to resume machine state here.

What do you think would be the better approach @krisjans @oliexdev ?

jensMF, 2021-12-07:
I read through the logs again, and some of the statements I made in my last post can't be true. Nevertheless, with the changes I did, it worked on three different devices where I had problems before. But if I do only some of the changes, it did not work.

Some things I forgot to mention: CHARACTERISTIC_CHANGE_INCREMENT has to be 4 bytes long and the birthday, too.

jensMF, 2021-12-09:
#806 should fix it

thomas725, 2024-03-26:
Hi there! I've recently upgraded from my Beurer BF-100 to the BF-105 and created a [modified version](https://gitlab.com/thomas351/beurer_bf100_parser/-/blob/bf105/src/main.rs) of my [BF-100 USB-data parser](https://gitlab.com/thomas351/beurer_bf100_parser) and now stumbled upon this project which seems to be able to read data from the BF-105 via bluetooth! nice! Though after having tried it seems to be missing the fat & muscle percentages for upper and lower body halfs, and instead only features the sum values for both. Is that a limitation of the bluetooth interface of the BF-105 or just me not having found it in the app or "just" the current implementation not reading / storing it?

Gerii, 2024-03-27:
It is in the Bluetooth logs: https://github.com/oliexdev/openScale/issues/635#issuecomment-818340540

So it just wasn't added because openScale is missing support to store and display upper/lower body values separately.