oliexdev/openScale #636: Document the import format

lgommans, 2020-10-03:
Took me forever to find out what format the import format is supposed to be. After finding in the [package description](https://f-droid.org/en/packages/com.health.openscale/) that it should be some form of CSV, the columns are not documented anywhere so eventually I dove into the source code ([1](https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/utils/CsvHelper.java), [2](https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/datatypes/ScaleMeasurement.java)) to find which columns it wants, the dateTime column turns out not to accept the ISO format (it wants some pseudo-ISO), headers are matched case-sensitive, the toast message specifying what the import issue is and what it should be instead disappears way too fast to read...

We should document what formats are accepted.

For reference, one format that I found works is this (mind casing):

```
dateTime,weight
2020-01-31 23:59,61.1
2020-12-31 13:59,60.0
```

oliexdev, 2020-10-12:
The date / time format is the ISO 8601:2004 see issue #506 

I also don't know why it took forever to find out the export format? It is written in the description and you could just simple do a sample export of the data and look into the CSV file? Sure you can improve the error handling or the documentation, just send a PR.

closed 2020-10-12 by oliexdev

lgommans, 2020-10-17:
I'm not sure why you closed the ticket as the issue, missing documentation, is not resolved. Is this not how you track tasks that remain to be done? Also, the label should not be "question": I already figured out and wrote above what the format needs to be; it's an "enhancement" for lack of a documentation-specific label.

> It is written in the description

The app description just says csv, not anything about columns or the formats thereof.

> you could just simple do a sample export of the data and look into the CSV file?

That's a good idea I didn't think of. Perhaps that could be the recommendation to put in the documentation?

> The date / time format is the ISO 8601:2004

Pretty sure that was my first try, but I don't have the original data file anymore to make sure. I typically assume it wants ISO first and then start to look at documentation or, failing that, source code. I would retry to see if there is an issue there but, y'know, there's one sure way to make sure people don't care enough to contribute to your project any further: close the unresolved ticket after questioning why they're having the problem in the first place.