oliexdev/openScale #637: Wrong muscle mass evaluation scale?

ThibG, 2020-10-07:
I have just bought a Mi Body Composition Scale 2 and while (except for visceral fat which is missing in openScale) the results are similar, their interpretations are pretty different, especially regarding the muscle evaluation scale.

Indeed, the measured muscle mass is ~40kg, which Mi Fit finds satisfying in a 36.5kg-42.6kg scale, while openScale finds it to be way too much, with an acceptable range of 20.6kg-23.7kg.

Is openScale's muscle mass evaluation off? Where do those ranges come from? Am I missing something?

ThibG, 2020-10-07:
I don't know how reliable those values are, but they are pretty different from those in openScale: https://support.withings.com/hc/en-us/articles/220035767-Body-Cardio-What-are-the-normal-ranges-for-body-composition-

https://github.com/oliexdev/openScale/blob/c790ee611dd3cd8650b37d8ee7b076c9dec7c616/android_app/app/src/main/java/com/health/openscale/core/evaluation/EvaluationSheet.java#L122-L139

oliexdev, 2020-10-12:
As you can see in the openScale source code an acceptable range between 20.6 and 23.7 kg didn't exist!?

Could you make a screenshot and post your entered age and sex or better add the openScale debug log file?

ThibG, 2020-10-12:
I don't have the exact values I used at the time right now (I remember I tried setting both genders and both gave evaluation results that seemed highly inconsistent with other apps) and I rebuilt openScale with my own values (from the articles I linked earlier), so I can't say for sure, but weighing around 60kg, the 20.6~23.7kg range seem consistent with the  34~39 percentage values for women aged 20-29 (which is the range I fall into) in the code linked earlier.

oliexdev, 2020-10-12:
I used the values from https://sanitas-online.de/media/download/751-737-1214_sbf70_de.pdf

I guess some scales use different definitions, as Mi scale uses skeletal, smooth, and cardiac muscle!? But I am not a medical expert.

ThibG, 2020-10-12:
Indeed, that may be down to differences across brands in what the muscle mass refers to… in Mi Fit's case and the corresponding MiScaleLib.java file, muscle mass is defined as total mass minus fat and bones, so this seems consistent with the higher ranges…

closed 2020-10-12 by oliexdev