oliexdev/openScale #638: pixel 4a crash at measurement with beta 2.3.1_dev

recyclerealestate, 2020-10-13:
open app.  start bluetooth icon on app, step on scale.  scale connects. scales does measurement. when it finishes(or when it sends data to phone) the app crashes "unexpected error".  it has worked once, in September...  looks like some sort of month error.  I don't know.

scale is a blue tooth "1 by one" ordered off Amazon.

______  error log below _____
Build version: 2.3.1-dev_1a75826d_2020-10-04 
Build date: 1981-01-01 01:01:02 
Current date: 2020-10-13 16:19:23 
Device: Google Pixel 4a 
 
Stack trace:  
java.lang.IllegalArgumentException: MONTH
	at java.util.GregorianCalendar.computeTime(GregorianCalendar.java:2662)
	at java.util.Calendar.updateTime(Calendar.java:3403)
	at java.util.Calendar.getTimeInMillis(Calendar.java:1761)
	at com.health.openscale.core.bluetooth.BluetoothOneByone.parseBytes(BluetoothOneByone.java:223)
	at com.health.openscale.core.bluetooth.BluetoothOneByone.onBluetoothNotify(BluetoothOneByone.java:125)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:358)
	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:443)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7656)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)

oliexdev, 2020-10-14:
strange does anybody else has the same error? Could you reinstall the app and try again?

recyclerealestate, 2020-10-14:
it is strange.  I did uninstall and re-install.  same behavior.  I
uninstalled everything then installed the previous version(the current one
on fdroid *Version 2.3.1* (50) added on 2020-08-21)      got this slightly
different error when it first connected to the scale:

Build version: 2.3.1
Build date: 1979-11-29 19:00:00
Current date: 2020-10-14 11:11:58
Device: Google Pixel 4a

Stack trace:
java.lang.NullPointerException: Attempt to invoke virtual method 'boolean
android.content.Context.isUiContext()' on a null object reference
at android.view.ViewConfiguration.get(ViewConfiguration.java:502)
at android.view.View.<init>(View.java:5317)
at android.view.View.<init>(View.java:5467)
at android.view.ViewGroup.<init>(ViewGroup.java:697)
at android.widget.LinearLayout.<init>(LinearLayout.java:254)
at android.widget.LinearLayout.<init>(LinearLayout.java:250)
at android.widget.LinearLayout.<init>(LinearLayout.java:246)
at android.widget.LinearLayout.<init>(LinearLayout.java:242)
at
com.health.openscale.gui.preferences.BluetoothSettingsFragment$BluetoothDeviceView.<init>(BluetoothSettingsFragment.java:264)
at
com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:191)
at
com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
at
com.health.openscale.gui.preferences.BluetoothSettingsFragment$1.onDiscoveredPeripheral(BluetoothSettingsFragment.java:129)
at com.welie.blessed.BluetoothCentral$2$1.run(BluetoothCentral.java:170)
at android.os.Handler.handleCallback(Handler.java:938)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loop(Looper.java:223)
at android.app.ActivityThread.main(ActivityThread.java:7656)
at java.lang.reflect.Method.invoke(Native Method)
at
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)

oliexdev, 2020-10-21:
sorry but I can't help you really. It seems an error in the Android library which is very strange.

recyclerealestate, 2020-10-21:
thanks for trying.  I think it is something to do with reading historical
weights...  I cleared cache, deleted everything which removed all my data
and it just now worked. no probably had data going back 2years or so.

anyway, seems like it is working now.

oliexdev, 2020-10-22:
ok even more strange those errors doesn't has to do with the database but ok

closed 2020-10-22 by oliexdev