oliexdev/openScale #639: Xiaomi scale does not synchronize.

Sulas85, 2020-10-15:
**Describe the bug**
When trying to synchronize my xiaomi mi scale 2 gives an error and does not let me continue. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Open the app for the first time.'
2. Click on 'Synchronize scale '
3. See error:

Build version: 2.3.1-pro 
Current date: 2020-10-07 11:35:34 
Device: HUAWEI VOG-L29 
 
Stack trace:  
java.lang.IllegalArgumentException: ID does not reference a View inside this Activity
	at android.app.Activity.requireViewById(Activity.java:3362)
	at androidx.core.app.ActivityCompat.requireViewById(ActivityCompat.java:363)
	at androidx.navigation.Navigation.findNavController(Navigation.java:58)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$BluetoothDeviceView.onClick(BluetoothSettingsFragment.java:355)
	at android.view.View.performClick(View.java:7213)
	at android.view.View.performClickInternal(View.java:7187)
	at android.view.View.access$3500(View.java:828)
	at android.view.View$PerformClick.run(View.java:27675)
	at android.os.Handler.handleCallback(Handler.java:900)
	at android.os.Handler.dispatchMessage(Handler.java:103)
	at android.os.Looper.loop(Looper.java:219)
	at android.app.ActivityThread.main(ActivityThread.java:8347)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1055)

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): ??? Yes

**Expected behavior**
A clear and concise description of what you expected to happen.

I need to be able to synchronize the scale 

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

Sulas85, 2020-10-15:
Debug log:

2020-10-15 13:26:23.534 Debug [1] AboutPreferences: Debug log enabled, openScale v2.3.1-pro (50), SDK 29, HUAWEI VOG-L29
2020-10-15 13:26:23.536 Debug [1] AboutPreferences: Selected user id(1) name(Jesús ) birthday(Thu Apr 18 00:00:00 GMT+02:00 1985) age(35) body height(172,00) scale unit(kg) gender(male) initial weight(79,10) goal weight(73,00) goal date(Tue Sep 28 00:00:00 GMT+02:00 2021) measure unt(cm) activity level(0) assisted weighing(true)

oliexdev, 2020-10-21:
could you try the latest dev build version?

Sulas85, 2020-10-21:
Hello, I tried the development version but the same thing happens.

oliexdev, 2020-11-23:
Unfortunately, I can't reproduce this error. Seems to be some context problem.

oliexdev, 2020-11-25:
should be fixed but it happened not during the synchronization but on selecting the scale via the slide show.

closed 2020-11-25 by oliexdev