oliexdev/openScale #643: Exception when connecting

alexanderadam, 2020-11-12:
**Describe the bug**
I had to reinstall openScale since I uninstalled openScale for testint the #641 build.
When I tried to connect with the scale I ran into an exception.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to connect to a scale

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 2.3.1 from F-Droid

**Expected behavior**
No error shown

**Additional context**
Nothing special

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

```
Build version: 2.3.1 
Build date: 1979-11-30 01:00:00 
Current date: 2020-11-12 06:14:46 
Device: Samsung SM-T860 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.res.Resources android.content.Context.getResources()' on a null object reference
	at android.view.ViewConfiguration.get(ViewConfiguration.java:491)
	at android.view.View.<init>(View.java:5432)
	at android.view.View.<init>(View.java:5624)
	at android.view.ViewGroup.<init>(ViewGroup.java:687)
	at android.widget.LinearLayout.<init>(LinearLayout.java:254)
	at android.widget.LinearLayout.<init>(LinearLayout.java:250)
	at android.widget.LinearLayout.<init>(LinearLayout.java:246)
	at android.widget.LinearLayout.<init>(LinearLayout.java:242)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$BluetoothDeviceView.<init>(BluetoothSettingsFragment.java:264)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:191)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1.onDiscoveredPeripheral(BluetoothSettingsFragment.java:129)
	at com.welie.blessed.BluetoothCentral$2$1.run(BluetoothCentral.java:170)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:237)
	at android.app.ActivityThread.main(ActivityThread.java:8167)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
```

oliexdev, 2020-11-23:
Did you edit some of the source-code because the line numbers of the errors seems really off.

Do you really tried to connect to the scale because the error seems to be in the Bluetooth settings menu.

alexanderadam, 2020-11-24:
> Did you edit some of the source-code because the line numbers of the errors seems really off.

No I didn't. I'm not an Android developer and the only time I ever checked out the project was for adding [a single line in the manifest.xml](https://github.com/alexanderadam/openScale/commit/7b1247335b7896f644893d8f824b79f65abc68d3).
And the exception happened in the F-Droid that I installed after uninstalled the modified build.

> Do you really tried to connect to the scale because the error seems to be in the Bluetooth settings menu.

I guess, but I cannot remember exactly any details (it was two weeks ago and I even forgot about this issue).

oliexdev, 2020-11-24:
did this error happened again with the latest f-droid version?

alexanderadam, 2020-11-24:
No, because, it also just happened once. And only on the first(!) connection (I think I forgot to mention this — sorry!).
So it was not the 'regular' connection thingy to sync data while the scale is already known. It was the first connection (the moment when openScale doesn't know about the scale yet).

The import/export of openScale only affects a single account, right?
So there's no possibility to setup all settings and users with a single export because the settings are lost and only health data for a single user is imported. Or do I misremember this? :thinking:

oliexdev, 2020-11-24:
> No, because, it also just happened once. And only on the first(!) connection (I think I forgot to mention this — sorry!).
> So it was not the 'regular' connection thingy to sync data while the scale [...]

Ok then I will close this issue here because the line numbers are really off. I can't reproduce this error neither.

> The import/export of openScale only affects a single account, right?
> So there's no possibility to setup all settings and users with a single export because the settings are lost and only health data for a single user is imported. Or do I misremember this? thinking

Yes, the CSV exports only the selected user but you can export the whole measurement database (including all users) in the settings->backup. The openScale and user settings are always lost but these are not so important and are quickly configured.

closed 2020-11-24 by oliexdev

alexanderadam, 2020-11-24:
> the line numbers are really off

Can you elaborate on that? At least the ones [in `BluetoothSettingsFragment.java`](https://github.com/oliexdev/openScale/blob/v2.3.1/android_app/app/src/main/java/com/health/openscale/gui/preferences/BluetoothSettingsFragment.java) seem to match (I didn't check the other parts, since they aren't within the `om.health.openscale` namespace).

oliexdev, 2020-11-24:
sorry you are right because it uses an old version of BluetoothSettingsFragement the latest version is a little bit different, so that's why the line numbers are off.

alexanderadam, 2020-11-24:
Does that mean, that the F-Droid uses old dependencies that might have caused this issue?

oliexdev, 2020-11-24:
No the latest version on f-droid is 2.3.2