oliexdev/openScale #645: Crashed during Startup

sheeeng, 2020-11-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): ???

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

```text
Build version: 2.3.2 
Build date: 1981-01-01 01:01:02 
Current date: 2020-11-15 12:11:08 
Device: Motorola moto g(7) power 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=4; index=4
	at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
	at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
	at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addPredictionLine(ChartMeasurementView.java:535)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addTrendLine(ChartMeasurementView.java:512)
	at com.health.openscale.gui.measurement.ChartMeasurementView.refreshMeasurementList(ChartMeasurementView.java:392)
	at com.health.openscale.gui.measurement.ChartMeasurementView.updateMeasurementList(ChartMeasurementView.java:347)
	at com.health.openscale.gui.overview.OverviewFragment.updateOnView(OverviewFragment.java:264)
	at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:239)
	at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:236)
	at androidx.lifecycle.LiveData.considerNotify(LiveData.java:131)
	at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:149)
	at androidx.lifecycle.LiveData.setValue(LiveData.java:307)
	at androidx.lifecycle.LiveData$1.run(LiveData.java:91)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:241)
	at android.app.ActivityThread.main(ActivityThread.java:7604)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:941)
```

davidhildebrand, 2020-11-16:
Experiencing a similar issue immediately upon opening the app.

`Build version: 2.3.2 
Build date: 1981-01-01 01:01:02 
Current date: 2020-11-14 09:44:44 
Device: Google Pixel 2 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=4; index=4
 at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
 at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
 at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
 at com.health.openscale.gui.measurement.ChartMeasurementView.addPredictionLine(ChartMeasurementView.java:535)
 at com.health.openscale.gui.measurement.ChartMeasurementView.addTrendLine(ChartMeasurementView.java:512)
 at com.health.openscale.gui.measurement.ChartMeasurementView.refreshMeasurementList(ChartMeasurementView.java:392)
 at com.health.openscale.gui.measurement.ChartMeasurementView.updateMeasurementList(ChartMeasurementView.java:347)
 at com.health.openscale.gui.overview.OverviewFragment.updateOnView(OverviewFragment.java:264)
 at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:239)
 at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:236)
 at androidx.lifecycle.LiveData.considerNotify(LiveData.java:131)
 at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:149)
 at androidx.lifecycle.LiveData.setValue(LiveData.java:307)
 at androidx.lifecycle.LiveData$1.run(LiveData.java:91)
 at android.os.Handler.handleCallback(Handler.java:938)
 at android.os.Handler.dispatchMessage(Handler.java:99)
 at android.os.Looper.loop(Looper.java:223)
 at android.app.ActivityThread.main(ActivityThread.java:7656)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)`

oliexdev, 2020-11-23:
Unfortunately I can't reproduce this error. How many measurements do you have? Did this error occur if you reinstall openScale?

sheeeng, 2020-11-23:
It happens on every measurements. The problems occurs upon startup even after removal and installation.

oliexdev, 2020-11-23:
That's strange could you paste the error log after a fresh installation of openScale?

sheeeng, 2020-11-23:
```text
Build version: 2.3.2 
Build date: 1981-01-01 01:01:02 
Current date: 2020-11-23 12:14:07 
Device: Motorola moto g(7) power 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=4; index=4
	at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
	at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
	at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addPredictionLine(ChartMeasurementView.java:535)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addTrendLine(ChartMeasurementView.java:512)
	at com.health.openscale.gui.measurement.ChartMeasurementView.refreshMeasurementList(ChartMeasurementView.java:392)
	at com.health.openscale.gui.measurement.ChartMeasurementView.updateMeasurementList(ChartMeasurementView.java:347)
	at com.health.openscale.gui.overview.OverviewFragment.updateOnView(OverviewFragment.java:264)
	at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:239)
	at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:236)
	at androidx.lifecycle.LiveData.considerNotify(LiveData.java:131)
	at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:149)
	at androidx.lifecycle.LiveData.setValue(LiveData.java:307)
	at androidx.lifecycle.LiveData$1.run(LiveData.java:91)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:241)
	at android.app.ActivityThread.main(ActivityThread.java:7604)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:941)
```

oliexdev, 2020-11-23:
That's somehow not possible this error because this method is not called during a fresh installation without any measurements!?
Did you really completely remove the openScale app on your phone and install it again?

sheeeng, 2020-11-23:
| Did you really completely remove the openScale app on your phone and install it again?
Yes.

davidhildebrand, 2020-11-23:
The error also persists for me.
I attempted to uninstall openScale and also remove the backups from internal storage.  Then re-installed with F-Droid.  Same error.
I had only taken and stored one measurement.  The app started showing this error when I went to take another measurement about a day later.

Is there anything else that should be removed or uninstalled before re-installing that could help?

Thanks!

sheeeng, 2020-11-23:
> The error also persists for me and happens immediately upon opening the app.
> I attempted to uninstall openScale and also remove the backups from internal storage. Then re-installed with F-Droid. Same error.
> I had only taken and stored one measurement. The app started showing this error when I went to take another measurement about a day later.

I have used the same steps.

oliexdev, 2020-11-23:
We need more details when this error occurred so please describe in detail
from a fresh installation every step.

2020, 22:01:

EddyTDev, 2020-11-24:
Same problem for me..... :'(
I just can't launch the program : 

`Build version: 2.3.2 
Build date: 1981-01-01 01:01:02 
Current date: 2020-11-23 20:01:35 
Device: Xiaomi Redmi Note 7 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=4; index=4
	at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
	at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
	at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addPredictionLine(ChartMeasurementView.java:535)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addTrendLine(ChartMeasurementView.java:512)
	at com.health.openscale.gui.measurement.ChartMeasurementView.refreshMeasurementList(ChartMeasurementView.java:392)
	at com.health.openscale.gui.measurement.ChartMeasurementView.updateMeasurementList(ChartMeasurementView.java:347)
	at com.health.openscale.gui.graph.GraphFragment$10.onChanged(GraphFragment.java:283)
	at com.health.openscale.gui.graph.GraphFragment$10.onChanged(GraphFragment.java:280)
	at androidx.lifecycle.LiveData.considerNotify(LiveData.java:131)
	at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:149)
	at androidx.lifecycle.LiveData.setValue(LiveData.java:307)
	at androidx.lifecycle.LiveData$1.run(LiveData.java:91)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:224)
	at android.app.ActivityThread.main(ActivityThread.java:7590)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
`

oliexdev, 2020-11-24:
And the same request

EddyTDev, 2020-11-24:
- I try to delete the cache of the application : same problem.

- I try to delete all my data for the app in android setting... (without reinstalling OpenScale)
OpenScale now open. I have lost all my data but I can re-use your program.

oliexdev, 2020-11-24:
> I try to delete all my data for the app in android setting... (without reinstalling OpenScale)

do you still know how many data you had?

> The error also persists for me and happens immediately upon opening the app.
> I had only taken and stored one measurement. 

The two statements are contradictory for me. How can you add a measurement if the app crash on startup? What kind of data was the measurement? Please enable the openScale log file if you can enter the app. Without more details I can't really help because I can't reproduce this error here.

oliexdev, 2020-11-25:
@all please try out the latest dev build version at https://github.com/oliexdev/openScale/releases/tag/travis-dev-build

kauzu, 2020-12-05:
I'm experiancing the same issue. 
I was using the app without any problems for about 2 weeks now with just a single user.
As we got a bluetooth scale today i wanted to try out if the app matches the weights of 2 different persons correctly. so i created a second profile in the app. from there on i did the following clicks:
1) go to table view of my profile ( with a few measurements)
2) go to graph view of my profile 
3) swicht to the new user (via this bar just above the graph.
Then it crashed. i guess this is due the new user not having any measurements so far

it also crashes right after i open the app, so it's unusable at the moment. i don't want to delete all date at this point as i dont want to lose it.

oliexdev, 2020-12-09:
@kauzu sure I can't still reproduce this error so please provide your data.

kauzu, 2020-12-10:
just send via mail. didn't publish here as i'm not 100% sure if there isn't any additional private data in the backup.

oliexdev, 2020-12-11:
thanks to @kauzu I was able to reproduce this error and to fix it. The problem was that your data had only few points that had the same x position.

closed 2020-12-11 by oliexdev