oliexdev/openScale #646: Exception when connecting (2.3.2)

eecummings, 2020-11-16:
**To Reproduce**
Step on scale, wait for connection.

**Additional context**
Problem began with 2.3.2 update.

**Debug log**
Build version: 2.3.2 
Build date: 1981-01-01 01:01:02 
Current date: 2020-11-15 07:35:10 
Device: LENOVO Lenovo TB-8705F 
OS version: Android 9 (SDK 28) 
 
Stack trace:  
java.lang.IllegalArgumentException: MONTH
	at java.util.GregorianCalendar.computeTime(GregorianCalendar.java:2662)
	at java.util.Calendar.updateTime(Calendar.java:3397)
	at java.util.Calendar.getTimeInMillis(Calendar.java:1761)
	at com.health.openscale.core.bluetooth.BluetoothOneByone.parseBytes(BluetoothOneByone.java:223)
	at com.health.openscale.core.bluetooth.BluetoothOneByone.onBluetoothNotify(BluetoothOneByone.java:125)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:358)
	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:461)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6714)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:503)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)

oliexdev, 2020-11-23:
could you attach the openScale debug file while you step on the scale.

oliexdev, 2020-11-25:
error should be catch now.

closed 2020-11-25 by oliexdev

vger76, 2020-11-29:
This issue is still relevant after 2.3.3 update: Build version: 2.3.3-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2020-11-29 03:36:26 
Device: Samsung SM-A715F 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.NullPointerException: no valid service UUID provided
	at java.util.Objects.requireNonNull(Objects.java:228)
	at com.welie.blessed.BluetoothPeripheral.getCharacteristic(BluetoothPeripheral.java:1077)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:200)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:188)
	at com.health.openscale.core.bluetooth.BluetoothSenssun.synchroniseDate(BluetoothSenssun.java:199)
	at com.health.openscale.core.bluetooth.BluetoothSenssun.onNextStep(BluetoothSenssun.java:85)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:460)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.access$000(BluetoothCommunication.java:46)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$2.onConnectedPeripheral(BluetoothCommunication.java:370)
	at com.welie.blessed.BluetoothCentral$4$1.run(BluetoothCentral.java:251)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:237)
	at android.app.ActivityThread.main(ActivityThread.java:8167)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)

Scale is Senssun/Grundig