oliexdev/openScale #647: Beurer BF700 ble protocol

MFoerd, 2020-11-17:
Hi,
I am currently trying to communicate with the Beurer BF700 via gatttool. The description that I found in your repo was very helpful here: [](https://github.com/oliexdev/openScale/wiki/Beurer-Sanitas
)
Unfortunately, I fail to initialize the scale and get no notification back when I request the status:

gatttool -I -b c8:b2:1e:c9:20:2d
[c8:b2:1e:c9:20:2d][LE]> connect 
Attempting to connect to c8:b2:1e:c9:20:2d
Connection successful
[c8:b2:1e:c9:20:2d][LE]> char-write-req 0x0021 0300
Characteristic value was written successfully
[c8:b2:1e:c9:20:2d][LE]> char-write-req 0x0021 e601
Characteristic value was written successfully
Notification handle = 0x0022 value: e6 00 20 
[c8:b2:1e:c9:20:2d][LE]> char-write-req 0x0021 e95FB381EC
Characteristic value was written successfully
[c8:b2:1e:c9:20:2d][LE]> char-write-req 0x0021 e74f0000000000000000
Characteristic value was written successfully
[c8:b2:1e:c9:20:2d][LE]> 

Do you have a hint for me what's wrong here? Any help would be great.

oliexdev, 2020-11-23:
No, I am sorry I never use the gatttool.

closed 2020-11-23 by oliexdev

MFoerd, 2020-11-23:
Thank you for your answer. In the meantime I have tried again and this behavior occurs when the scale has not been paired. Bluetoothctl turned out to be the better tool and I can now follow the valuable protocol of your description.