oliexdev/openScale #648: Renpho ES-CS20M not found

giupardeb, 2020-11-17:
I bought a Renpho ES-CS20M because it is supported, but when I can't finding the scale through the app (with blue-tooth and Location enabled). Why?

oliexdev, 2020-11-23:
What do you mean you can't find the scale? You can't see the scale at all in Android? You can't see while you search in openScale in the settings?

giupardeb, 2020-11-23:
I can't see the scale when I searching it in openScale. With the official application ([renpho](https://play.google.com/store/apps/details?id=com.qingniu.renpho&hl=en_US&gl=US)), I can find the scale.

oliexdev, 2020-11-23:
Then you probably has paired the Bluetooth scale with your phone already. Please uninstall the official app and try again.

giupardeb, 2020-11-24:
i tried but it keeps getting the same problem. Openscale Does not found the scale

oliexdev, 2020-11-24:
reset your scale and unpair the scale in the Android system. This only happens if the scale is already connected or not on.

closed 2020-11-24 by oliexdev

GreyMarvel, 2020-12-02:
I have the same issue , in my phone's Bluetooth menu I can see the scales Bluetooth address when I stand on them but in the openscale app it doesn't discover the renpho scales . Is it possible they have changed something in the scales firmware on newer versions of this model ?