oliexdev/openScale #651: Backup is not including latest data

tjaderxyz, 2020-11-19:
**Describe the bug**
I have daily backup enabled, but for quite some time now it seems to be doing the backup daily as expected but the exported file does not have the latest data.

**To Reproduce**
Steps to reproduce the behavior:
1. Have daily backup enabled
2. Wait a few days
3. See that the backup file is being written every day but contains stale data

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): no

**Expected behavior**
Daily backup should always include up to date data.

**Additional context**
Here's what I saw in adb shell this morning:
```
OnePlus5T:/ $ stat -c %y "$OSBACKUP"; sqlite3 "$OSBACKUP" 'SELECT strftime("%Y-%m-%d", MAX(datetime)/1000, "unixepoch") FROM scaleMeasurements;'                                                                                                                                         
2020-11-18 15:15:20.774253164 -0300
2020-11-15
```

The modification time in the backup file is 2020-11-18 15:15, but the most recent data in it is from 2020-11-15 even though I have several measurements after that:
[image]

After this I went in the backup menu and disable/reenabled daily backup and now I get this:
```
OnePlus5T:/ $ stat -c %y "$OSBACKUP"; sqlite3 "$OSBACKUP" 'SELECT strftime("%Y-%m-%d", MAX(datetime)/1000, "unixepoch") FROM scaleMeasurements;'                                                                                                                                         
2020-11-19 12:13:23.657720665 -0300
2020-11-19
```

**Debug log**
I enabled debug log now, can provide it in a few days if necessary.

oliexdev, 2020-11-23:
What happened if you manually export the database? Are then always the latest measurements included?

tjaderxyz, 2020-11-23:
Yes, when I manually export the database it includes all data.

oliexdev, 2020-11-23:
Ok the export method works correct.

Maybe we find something in the openScale log file, could you attach it here?

On Android >= 29 the permissions to write a file on the external storage is more restrictive. Maybe we have a problem here. Currently, I use the workaround to enable the requestLegacyExternalStorage option in the AndroidManifest

tjaderxyz, 2020-11-23:
It's hard to get the log file because it seems to only write to the log file while the application is open. If I leave the app and come back the logfile checkbox is unchecked again. I just left the app open for a couple of hours until it did the daily backup, but this time it included the most recent data.

Is there something else I can do to help investigate? Leaving the app open for days at a time to get the log is not viable.

oliexdev, 2020-11-23:
Indeed the app must be open to write the log file. It seems more an issue/feature of the Android system itself. The recent releases of Android use for battery saving some optimization. So background task are handled more restrictive by Android in every update. You could try to disable all battery saver features on your Android system and see if it works better.

closed 2020-11-23 by oliexdev

tjaderxyz, 2020-11-23:
I don't think it's that, because the file's modification time does get updated every day. That means Android must be waking up the app and the app is touching the file, but for some reason it's not doing everything it should.

oliexdev, 2020-11-23:
strange but then we need some log files from this event.

tjaderxyz, 2020-11-23:
I'm willing to help if I don't have to keep the app open for days in my
phone. Maybe it could log to logcat instead of a file?

oliexdev, 2020-11-24:
Just run it in Android Studio all logs are written via logcat in debug modus.