oliexdev/openScale #655: Inconsistent Measurements

mzzntn, 2020-11-22:
**Describe the bug**
Hi I recently buy a new scale 

I moved from a generic Kamtron #373 to a Xiaomi MiScale Body composition 2  I checked the compatibility as mentioned in #650 but the measurement from the two scale are completely different.

[image]

As you can see in the picture I gained 20kg of muscle and 5 of fat (in 2 days (I know that the measurements aren't completed at the same time but it shouldn't make too much difference) 

I suppose that the measurements from the Kamtron are more real but I actually have no really idea at this point :)  

I just want to understand which measurement are more accurate and stick with them (Even though would be a shame lose one month of measurements) In the really I actually have not idea if it's a scale, app or other error.

**Additional context**
Unfortunately I can't measure anymore with the old scale (I changed home)

mzzntn, 2020-11-22:
I just noticed that in #637 #543 #412 similar problems are mentioned

oliexdev, 2020-11-23:
so the other values are from the Kamtron scale?

mzzntn, 2020-11-23:
Yes, from the 29/10 to 20/11

oliexdev, 2020-11-23:
then it could be possible that the mi scale use some different definitions of the muscle mass and so on. You can't compare different scales from different vendors.

closed 2020-11-23 by oliexdev