oliexdev/openScale #656: A menu is missing the dark theme

naoliv, 2020-11-26:
**Describe the bug**
It seems that one menu is missing the dark theme.

**To Reproduce**
When using the dark theme and clicking at the  **⋮**  menu:

[image]

It's possible to see that the menu isn't properly themed (it's using a light theme):

[image]

Reproduced with version `2.3.3-light`

PavelGorbanj, 2021-02-17:
if I chose Light scheme, I see the same 3 items. @oliexdev, should this issue be closed?

naoliv, 2021-02-17:
> if I chose Light scheme, I see the same 3 items. @oliexdev, should this issue be closed?

Why sould it be closed?
The menu should be dark when using the dark theme, like this one:

[image]

oliexdev, 2021-02-17:
@naoliv then fix it?

naoliv, 2021-02-17:
> @naoliv then fix it?

I really wish I knew how to fix this and other issues.

oliexdev, 2021-02-17:
you could learn it, that's why it tagged as "good first issue". I would appreciate it.

oliexdev, 2023-02-11:
should be fixed in the latest dev build

closed 2023-02-11 by oliexdev